bigtreetech/BIGTREETECH-TouchScreenFirmware #2135: [BUG] TFT 43 Screen goes white after 2-4 hours printing.

LTSpenny, 2021-08-21:
Screen turns white and unresponsive to touch controls. Nob responds by turning on led or timing led out. Have updated firmware to latest release. 

### Steps to reproduce

1. Turn on printer
2. Start a print from either TFT or onboard SD. 
3. Dont touch tft for 2-4 hours of printing

**Expected behavior**
Screen to remain functional

**Actual behavior**
Screen turns white and half dim. Unresponsive to any touch controls. 
### Hardware Variant

TFT43V3
### TFT Firmware Version & Main Board Firmware details
Marlin 2.0 and latest TFT release. TFT43_V3.0.27.x

[image]

https://drive.google.com/drive/u/3/mobile/folders/148tLwug30xQxUPRNkjg_y2aSF7J0-hRg?sort=13&direction=a

LTSpenny, 2021-08-21:
[image]
Nothing appears burned out on board.

LTSpenny, 2021-08-22:
Looking through reddit comments someone stated that it was possible the skr2 may not be providing enough power? This just seems odd as the screen can be fine for a few hours and then randomly turns white. The only new thing Ive done to my configuration is change the mosfet from the standard ender 5 plus creality one to a th3dstudio mosfet for better safety. 

Am i able to use the gnd and 5v pin to power the board while in use to test this out? Would i remove the 5v pins from the tft connector as well as exp 1? How does this tft dictate where it is getting its power from? As exp1 and the rs232 pins both have 5v in’s

oldman4U, 2021-08-22:
Hi.

This is not a user group, but the BTT TFT firmware repository. Do you expect a software bug to cause this issue?

LTSpenny, 2021-08-22:
@oldman4U im quite new to this tech so ill take the “i don't know” answer. If your saying that this is hardware ill just close it and order a new one. But at least there will be a search history so someone else with the same issue can find this. Which i can infer from your reply. Thanks.

closed 2021-08-22 by LTSpenny

oldman4U, 2021-08-22:
It is very likely a hardware issue. If it is the SKR or the TFT you can probably find out by replacing the 5V and GND of the serial cable with an old USB cable like described in the troubleshooting section.

Happy printing

LTSpenny, 2021-08-22:
@oldman4U thank you for the guidance. If i can buy you a coffee for your time sometime let me know. I appreciate what you and some of the other regulars do here.

LTSpenny, 2021-08-24:
@oldman4U so after a day of checking it and trying to figure out if it was hardware, it seems and im only thinking it was caused by one of two things. An old configuration, and in one of my configs where it states the brightness levels to set them on at 100 brightness. These where the only two things i had done different the past few days and i had just flashed it with new firmware (has bug that ignores comments in start gcode) and i think i uploaded the old config. 

For anyone else that finds this, redownlowding current config file fixed this for me. I also lowered the brightness to 90%. Dont know if that part helped as i haven't had a chance to troubleshoot that much.

oldman4U, 2021-08-24:
Hi

Seems your SKR can not deliver enough power under all circumstances. Using power from an external source could proof this theory.