bigtreetech/BIGTREETECH-TouchScreenFirmware #2137: [BUG] Skipping gcode commands

flynny75, 2021-08-23:
### Description
Gcode parser is skipping commands containing comments

### Steps to reproduce
Generate gcode with comments.

**Expected behavior**
Print should complete as before

**Actual behavior**
Nothing might happen, or printer is very unpredictable

### Hardware Variant
TFT70

### TFT Firmware Version & Main Board Firmware details
No problem with firmware binaries from commit e1a0ba1d2e38575b0c4056c553e0776f79defa64
Problem on next binaries commit on 185f3fea86e081d14d93b032e54ebaf70a41f8ea

### Additional Information
Problems with gcode as small as
`M140 S60 ; set bed temperature
;TYPE:Custom
; custom gcode: start_gcode
G28 ; home
G29 ; abl
;M420 S1 ; load last saved mesh (save with M500)
M206 Z0 ; reset z home offset (PETG can change)
; custom gcode end: start_gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance
; custom gcode: start_filament_gcode
; Filament gcode
; custom gcode end: start_filament_gcode
M109 S215 ; set temperature and wait for it to be reached
M190 S60 ; set bed temperature and wait for it to be reached`

kisslorand, 2021-08-23:
Did you bother to check #2129 and #2131?

oldman4U, 2021-08-25:
PR#2131 fixes it and has been merged into Master

Please close

@Msq001

closed 2021-08-25 by bigtreetech