bigtreetech/BIGTREETECH-TouchScreenFirmware #2139: [BUG] (short description)  error  tft35 gd  filament load menu

poumsoon, 2021-08-25:
When using the filament load function in the tft35 gd version, the warning window in the figure repeats infinitely.
Always press reset and start again.
Actually, the filament load button cannot be used. It seems that the function of mm units in the previous version was much more convenient. This function was useful when checking whether the extruder's step value was set correctly.

[image]

digant73, 2021-08-25:
check on Marlin fw that all the following settings are active:

# General options:
#   EEPROM_SETTINGS (in Configuration.h)
#   BABYSTEPPING (in Configuration_adv.h)
#   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
#   AUTO_REPORT_POSITION (in Configuration_adv.h)
#   M115_GEOMETRY_REPORT (in Configuration_adv.h)
#   M114_DETAIL (in Configuration_adv.h)
#   REPORT_FAN_CHANGE (in Configuration_adv.h)
#
# Options to support printing from onboard SD:
#   SDSUPPORT (in Configuration.h)
#   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)
#
# Options to support dialog with host:
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#   HOST_PROMPT_SUPPORT (in Configuration_adv.h)
#
# Options to support M600 with host & (Un)Load menu:
#   Options to support dialog with host (as pre requisite)
#   NOZZLE_PARK_FEATURE (in Configuration.h)
#   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
#   PARK_HEAD_ON_PAUSE (in Configuration_adv.h)
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)

The Extrude menu is also still present under the Menu->Movement->Extrude

oldman4U, 2021-08-30:
@Msq001 

Please close this ticket. User does not reply anymore and the problem reported is most likely not firmware related.

Thank you

closed 2021-09-01 by bigtreetech