bigtreetech/BIGTREETECH-TouchScreenFirmware #2142: [Q] HOST_PAUSE_M76 should be enabled or not

sarvenn, 2021-08-28:
While I was updating my Marlin fw, I noticed that a new prompt line is added to host command actions section.

#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  //#define HOST_PAUSE_M76
  #define HOST_PROMPT_SUPPORT
  //#define HOST_START_MENU_ITEM  // Add a menu item that tells the host to start
#endif

Should I enable the //#define HOST_PAUSE_M76 line or not to work properly with my TFT35 E3 V3.

Note: I have enabled pause gcode in my config.ini. See attached file.
[config_a.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7070903/config_a.zip)

digant73, 2021-08-28:
No. It is not needed and not used by the TFT

sarvenn, 2021-08-29:
Thnx Antonino.

closed 2021-08-29 by sarvenn