bigtreetech/BIGTREETECH-TouchScreenFirmware #2143: [Q] Can't Find RRF Macros List

Wombat37, 2021-08-29:
This question relates to the Macro button on my BTT-TFT35-E3 v3.0 driving a BTT-E3-RFF v1.1 board

I've just updated the TFT with the latest firmware and when I press the Macros button, I'm taken to the Custom list. I can't find the list of Macros anywhere.

I think this is related to a recent commit from PR#2136.

I see from this commit, that the RRF reference in the config_rrf.ini has been removed and the TFT FW knows that I'm running RRF from some other reference.

I've compared the two ini files config.ini and config_rrf.ini and they now appear to be identical. There's nothing to indicate that I'm running RRF. How does the TFT now know I'm running RRF (and enable the macro list)? If it's in the bin files, then we'll need different bin files for Marlin and RRF.

Is there some other setting I need to adjust to bring back the Macro list or is this a bug?

Wombat37, 2021-08-29:
I forgot to mention that I'm running RFF 3.3.

Wombat37, 2021-08-30:
Just installed newest firmware release with PR #2140.

Macro button now lists the macros!

All is well with the world.

I'll close this issue.

closed 2021-08-30 by Wombat37

pfn, 2021-08-30:
Yeah, I overlooked a setting when I had taken out the auto-detection. It worked fine if you upgraded from an older version, or something.