bigtreetech/BIGTREETECH-TouchScreenFirmware #2144: Shadow characters in Layer-field

brutzler, 2021-08-29:
[image]

### Description

When you printed a part with layer hight more than 10mm, and after this you start a new print, then the field where the actual layer ist displayed, still has the last (longer string) in background.

Its no such big bug. See it as feedback to enhance your software :-) 

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

BIGTREETECH_TFT35_V3.0
### TFT Firmware Version & Main Board Firmware details

TFT35_V3.0.27.x Aug 22 2021
Marlin 2.0.9.1

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

kisslorand, 2021-09-01:
@brutzler
Check please PR #2152?

brutzler, 2021-09-06:
Still on vacation for 2 more weeks....
Will test after back home again.
Silly question: Is there somewhere the file with the changes from you? Or do I have to change my sources?

PR #2152 (2021-09-07): Bugfix - shadow characters in Layer-field
### Requirements

BTT or MKS TFT

### Description

This PR adds more variables to be initialized before the start of a print.

### Related Issues

Fixes #2144
Adds a missing print loop before displaying the layer number.

### Notes

~~This PR is marked as draft because I am on vacation, far away from any 3D printer and have no way to test it. I hope someone will have the good will to test it and report back.~~
closed 2021-09-07 by bigtreetech