bigtreetech/BIGTREETECH-TouchScreenFirmware #2151: [FR] Scrolling with the encoder

bttguy, 2021-08-31:
**Is your feature request related to a problem? Please describe.**
Navigation in the files or in the list settings require multiple taps with a lot of option/gcode/macros.

**Describe the solution you'd like**
Would it be possible to use the encoder to scroll up and down/left or right in grid view/ as an option that can be configured in the UI settings/config.ini(to let people chose to turn on)? This would save a lot of time, and prevent unnecessary taps on the screen.

**Additional context**
This would be useful for all user, regadless the firmware.

oldman4U, 2021-09-01:
Feels a bit like this to me...
[image]

bttguy, 2021-09-01:
Well, I don't think it's the case, but I can accept that this not for you.
If we compare the encoder to an stylus why do we implemented temperature/fan/speed/flow change via encoder? Because it require multiple taps.
If there are 50+gcode and a lot of macros I am see a point here, especially when you get used to your file system any know what to look where/how far in the list. 
But this is still just my idea and preference, hopefully it will be implemented one day, and at least one more user will find it useful not just me :)

guruathwal, 2021-09-02:
@oldman4U any new feeling about this? 😂🤣

[image]

guruathwal, 2021-09-02:
@bttguy The main issue in adding full encoder support in touch UI is related to this one: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/680.
This issue is only partially resolved for boards that have both EXP1 and EXP2 headers that run Marlin firmware only.

oldman4U, 2021-09-02:
> @oldman4U any new feeling about this? 😂🤣
> 
> 
> 
> [image]
> 
> 

You know this is another use case my friend. Nobody operates his iPhone or iPad using a stylus. But maybe we could allow to paint with the encoder 🤪

bttguy, 2021-09-02:
> @bttguy The main issue in adding full encoder support in touch UI is related to this one: #680.
> This issue is only partially resolved for boards that have both EXP1 and EXP2 headers that run Marlin firmware only.

Hmm, yeah I saw this, but thought it has been resolved. Than RRF users will welcome this change without messing up Marlin users, we dont have to worry about this change. Since we can use unlimited numebr of macros and print anyway from the onboard SD's full library, not just what we copy to our flashdrive, this feature has more sense for us, but hoped it could be a general improvement. Any possibility to be added this or any way? @bigtreetech maybe you can help here to decide/code too. As a RRF user with your TFT and mainboard this is a vaild usecase with this firmware imo.
@oldman4U I would be also glad if we could implement some minigames like pong with the encoder, while we wait oir prints :)

oldman4U, 2021-09-02:
Tetris

gordo3di, 2021-09-03:
> Tetris

Tetris that auto completes as the print progress increases!

digant73, 2021-09-03:
@guruathwal as done on other menus (e.g. Heat), we could add the support for the navigation (rotating the knob). We know that the knob button MUST not be clicked. But at least the navigation will be more easy than pressing on the prev/next buttons. We will simply need to touch (press) on the screen to apply the action to the selected item

bttguy, 2021-09-03:
This would be awesome @digant73. Could you implement it as well in the future or we should seek for a dev with the possibility to make this work?

closed 2021-09-09 by bigtreetech