bigtreetech/BIGTREETECH-TouchScreenFirmware #2158: [BUG] Sounds settings in config.ini are not taken into account

GregSKR, 2021-09-03:
Looks like the sounds settings in **config.ini** are not taken into account with latest build...

e.g. with:
```
touch_sound:0
toast_sound:1
alert_sound:1
heater_sound:1
```
touch sound is not disabled

or with:
```
touch_sound:0
toast_sound:0
alert_sound:0
heater_sound:0
```
only touch_sound is disabled, not the others.

Can someone reproduce the bug?

kisslorand, 2021-09-04:
What about setting the sounds on the TFT? Does the settings there reflect the congig.ini settings? Does the settings on TFT reflect the sounds behaviour?

digant73, 2021-09-04:
See in the list of #2146 (it was a bug present also on other params). The bug was fixed in #2146 merged two days ago. You must load again the config.ini

kisslorand, 2021-09-04:
OMG! I was scratching my head for two hours now, I couldn't find anything wrong in the current master. :)

GregSKR, 2021-09-04:
Oups, my bad I thought I was on the latest build but I was not...
Just flashed 09/02 build and all is fine now.

Thanks for your answers.
Closing this ticket

closed 2021-09-04 by GregSKR