bigtreetech/BIGTREETECH-TouchScreenFirmware #2169: [FR] (Notify if Runout is Disabled)

gordo3di, 2021-09-09:
**Is your feature request related to a problem? Please describe.**
Sometimes we turn off the runout sensor for a variety of reasons. If we forget to re-enable it for the next print we might lose the print. Especially when several people are using the machines.

**Describe the solution you'd like**
Just have a popup notification display when a print is started that the runout sensor is turned off.

**Describe alternatives you've considered**
Yell at my team to stop turning off the sensor. Didn't work plus sometimes we have to bypass it.

I understand that not all printers have sensors so I'm wondering how to differentiate those machines vs ones where it's just "disabled". I added it to the top of PrintingMenu.c so it only appears once but I'm not sure if this is the correct spot.

I was thinking something as simple as:

```
void menuBeforePrinting(void)
{

//notify if runout is off
if (infoSettings.runout == 0)
{
addNotification(DIALOG_TYPE_ALERT, "Runout is OFF", "Runout Sensor is Turned OFF", true);
ackPopupInfo("Runout Sensor is OFF");
}
```

gordo3di, 2021-09-22:
Does anyone think this is worth implementing? Otherwise I'll close the ticket.

kisslorand, 2021-09-23:
I think it would be very unpopular among people who have the runout sensor connected to the motherboard or no runout sensor at all.

gordo3di, 2021-09-23:
> I think it would be very unpopular among people who have the runout sensor connected to the motherboard or no runout sensor at all.

Yeah. Without a way to differentiate off vs never installed it would be annoying for a lot of people.

closed 2021-09-23 by gordo3di