bigtreetech/BIGTREETECH-TouchScreenFirmware #2173: [FR] (Can we get IDEX support in the TFT firmware)

muttstang, 2021-09-13:
**Is your feature request related to a problem? Please describe.**
It appears that there is no IDEX support with the TFT firmware.

**Describe the solution you'd like**
Can we get a solution for duplication, mirror, dual extrusion along with moving the extruder head, moving filament in E1, and adjusting offsets for the 2 print heads.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

thisiskeithb, 2021-09-13:
It has already been requested: #2020

digant73, 2021-09-16:
@muttstang as said by @thisiskeithb, your FR is a duplication of #2020. Please close this ticket and eventually post on #2020

closed 2021-11-22 by stale[bot]