bigtreetech/BIGTREETECH-TouchScreenFirmware #2176: TFT-70 does not work with Marlin 2.0.9.1

Decalis1979, 2021-09-14:
I have two problems:
1- In marlin 2.0.9.1 I can't print directly from the SD card (BTT GTR V1.0) onboard gives me the following error on the display (cold extrusion prevented).
2- In marlin 2.0.9.1 I can't use the Marlin menu. Only the Bigtreetech menu works.

oldman4U, 2021-09-14:
I have two question:
1- Why do you open a ticket AND a discussion in the BTT TFT FIRMWARE repository, for issues which are most likely not related to the BTT TFT firmware?
2- Any hint, that the issues you describe are related to the TFT firmware?

Thank you

Decalis1979, 2021-09-14:
Previously on marlin 2.0.8.1 it worked perfectly, when upgrading to marlin 2.0.9.1 the marlin menu stopped working only the bigtreetech menu works. Why am I asking for help?

oldman4U, 2021-09-15:
So you have a good working solution consisting of two elements. Then it stops working after changing one of the two elements but you blame the one you did not change to be the reason that it does not work anymore!!?? Not sure I am able to follow this kind of logic.

Decalis1979, 2021-09-15:
Problema resolvido.

closed 2021-09-15 by Decalis1979