bigtreetech/BIGTREETECH-TouchScreenFirmware #2180: [FR]

credomane, 2021-09-17:
**Is your feature request related to a problem? Please describe.**
I suppose my problem would be that If I start a print job from OctoPrint then the BTT TFT doesn't get the "hint" and switch to the printing screen.

**Describe the solution you'd like**
I poked around the source came up with a super complicated solution that's way over my head...I put in the alternatives I considered question below. My actual solution was to just add some gcode to octoprint's "gcode scripts" settings and a small patch to the tft's source code.

P2 is just the serial port that the tft is attached to. `P2` isn't required and I could let it go out all serial ports but I didn't want to risk octoprint getting confused.

Octoprint start `M118 A1 P2 action:printstart`  
Octoprint complete `M118 A1 P2 action:printcomplete`  
Octoprint canel `M118 A1 P2 action:cancel`  
Octoprint pause `M118 A1 P2 action:paused`  
Octoprint pause `M118 A1 P2 action:resumed`  

Then in the BTT TFT source code I added this small patch:

```diff
diff --git a/TFT/src/User/API/parseACK.c b/TFT/src/User/API/parseACK.c
index 850dadfd..dfd639ad 100644
--- a/TFT/src/User/API/parseACK.c
+++ b/TFT/src/User/API/parseACK.c
@@ -304,6 +304,14 @@ void hostActionCommands(void)
       }
     }
   }
+  else if (ack_seen(":printstart"))
+  {
+    setPrintHost(true);//Think this is the right function based on code comments. Triggers the printing screen like I want.
+  }
+  else if (ack_seen(":printcomplete"))
+  {
+    printComplete();
+  }
   else if (ack_seen(":paused") || ack_seen(":pause"))
   {
     if (infoMachineSettings.firmwareType == FW_MARLIN)
```

**Describe alternatives you've considered**
Writing a LCD driver in marlin to handle serial comms between marlin and btt tft. That would require a significant rewrite of the current serial code in this repo. The plus side would be "Marlin Mode" would, in theory, work over serial and support touch (faked encoder messages). Left and top of screen would be counter clockwise, right and bottom clockwise, and center is a click. Complicated right? That's where my head went first. I even have a couple pages for the basics of the serial protocol. 

**Additional context**
So far this seems to be working with minimal issues. The one issue being that after a print completes the top left of the main screen displays `:?` where normally is says `Ready`. The print summary popup has the same `:?` in it for the "title"  
I'm sure it is something I'm not handling correctly. Input on that would be great.

If desired I could eventually turn this into a PR once all the bugs are worked out.

digant73, 2021-09-17:
In order to be merged on TFT fw, this protocol should be part of Octoprint or one of its plugins.

kisslorand, 2021-09-17:
I would also use M75 and M77 to tell the TFT the print has been started or stopped.

oldman4U, 2021-09-19:
@credomane 

Please reply to your own ticket. 

Thank you

credomane, 2021-09-20:
> @credomane
> Please reply to your own ticket.

This is a feature request for a way for the tft to detect externally started prints and I provided a badly thought out hacky partial wworkaround/solution. I don't think there is anything else of value for me to add, sadly. I'll look at the source some more this week and see if I can come up with something better but it isn't looking good.

oldman4U, 2021-09-20:
Happy coding. 👍🏻

lightmaster, 2021-09-22:
IIRC, Marlin has the ability to send out those Action Commands automatically when the events occur, ie when a print starts, is paused, is complete, etc. I know I've seen the notification prompts on the TFT, before I switch to the Marlin emulator mode when printing. I'll double check later today when I start another print.

Pretty sure its these lines from `Configuration_adv.h` that enable it.
https://github.com/MarlinFirmware/Marlin/blob/11ceea25ce145c8485b516fdb010809409d2bb06/Marlin/Configuration_adv.h#L3836-L3841
```
#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  //#define HOST_PAUSE_M76
  #define HOST_PROMPT_SUPPORT
  //#define HOST_START_MENU_ITEM  // Add a menu item that tells the host to start
#endif
```

digant73, 2021-09-23:
yes, the settings required in Marlin to enable host actions towards the TFT are also reported on top of the config.ini file.
If Octoprint already sends gcodes to Marlin that will be triggered by Marlin as host actions then I will simply need to know those action texts (e.g. //action:start etc...). Some actions notified when printing from onboard SD are already supported such as //action:pause, //action:resume, //action:cancel. Basically only the "start" and "stop" notifications are missing. If you are able to provide the list of actions triggered by Marlin I can simply add, in the TFT fw, the missing ones.

lightmaster, 2021-09-23:
So it does not say anything when paused or resumed, not sure why it's sending out `Printing...` when it starts. Perhaps something to do with the Layer Progress plugin.

lightmaster, 2021-09-23:
Also noticed that when it's paused via Octoprint, the Marlin emulated screen still has a pause option and doesn't realize it's actually paused.

digant73, 2021-09-23:
> So it does not say anything when paused or resumed, not sure why it's sending out `Printing...` when it starts. Perhaps something to do with the Layer Progress plugin.

what's displayed on TFT are //action:notification XXXX. That could be also OK to intercept the start/stop of a print. Pause/resume are currently intercepted by //action:pause and //action:resume. So they are not notifications displayed on the TFT. However this is not important for you (they are simply implementation details). I need to know the actions (notifications or anything specific) that are triggered by Marlin to TFT. E.g.:
//action:notification Printing (for start)
//action:notification Terminated (for stop)
//action:pause (for pause)
etc...
If you (anyone using Octoprint) could provide that info, it will be very easy to enhance the TFT fw to intercept those triggers

tekknix, 2021-09-23:
Currently ending a print, will try to collect the Informations afterwards.

tekknix, 2021-09-23:
Indeed this is language depending.

Now i've got for a Print-Start from Octoprint: 

//action:notification Druckt...
//action:notification Bett heizt...

But German-Language is not completely used.

//action:notification Bed leveling done.

tekknix, 2021-09-24:
It seems very related to this one: #1743 

Everyone "hunts" for the same. Start a Print from Octoprint (or any other "Host") and let the TFT understand that a Print-Job has started. I'm not really sure if this deals together with "Not SD printing" in ParseACK.c

For my understanding as long as Marlin is sending "Not SD printing" everythink regarding setPrintHost is set to false and no Print is detected. For sure with M117 and DisplayLayerProgressPlugIn we can make it work that Layer-Informations and more are sent to the Status-Page but still you can't interact with the TFT itself anymore. Even no Cancel, No Pause, ...

@credomane has tried to "force" a setPrintHost(true); do to parsing a special action from Octoprint starts, that the TFT is "detecting" that a print is running. 

Any thoughts?

digant73, 2021-11-21:
Once PR #2241 is merged, please test it and open a new thread in case of further enhancement requests or bug reports

closed 2021-11-25 by bigtreetech