bigtreetech/BIGTREETECH-TouchScreenFirmware #2181: [BUG] Some pop-ups softlock the TFT.

credomane, 2021-09-17:
### Description

So far I've managed to softlock the tft in two ways. First way is the Filament load/unload screen and the 2nd way was while I was doing a SD print from the tft's sdcard slot.

### Steps to reproduce

Filament unload:  
1. Warm up the hotend.
2. Enter the filament load/unload screen
3. Click load filament.
4. Get popup "Load more filament" with ok/cancel buttons.
5. Click "cancel"
6. get "Printer is paused. Continue?" with ok button.
7. Click "OK" to continue.
8. Jump to step 6 or skip to 9.
9. Hit tft reset button.
10. Jump to step 6 once tft reboots or goto 11.
11. Turn printer off and on again to restore functionality.

Fullscreen popups prevent resolving the reason for their appearance.

1. Start a sdcard print from the tft's sd card.
2. Print runs as normal.
3. This step I dunno how or what happened but roughly 1h into the print all the heaters turn *off*. There is no reason that I can find that this should have happened.
4. Once the extruder drops below the "Cold Extrusion Threshold". The TFT goes bananas.
5. Buzzer is blaring repeatedly. Pop up "Cold Extrusion Prevented" pops continues to appear as soon as you close it off.
6. You can't do crap because of 5. Can't abort the print, can't pause, can't turn the heaters back on. Only option is to reset the TFT.

**Expected behavior**
I don't even understand how/why the printer got into the "paused" state by loading filament  but anyways I fully expect clicking ok on the "continue" popup should, well, unpause the printer.

Would be nice if the "full screen" pops up had a cooldown of a 3-5 seconds before another one could be displayed. Would at least give me a few seconds to attempt to do something to resolve the situation.

**Actual behavior**
This got slipped into the steps to reproduce.

### Hardware Variant
BTT TFT70 V3 running BIGTREE_TFT70_V3.0.27.x.bin downloaded from this repo Sept 13th.
BTT Octopus V1.1 running Marlin bugfix-2.0.x compiled by me Sept 8th.

digant73, 2021-09-17:
for the first problem (load/unload) verify the Marlin settings reported on top of config.ini file are set on your Marlin fw

oldman4U, 2021-09-17:
You have seen?
9: Hit tft reset button.

oldman4U, 2021-09-19:
@credomane 

What is the status of the wrong Marlin configuration and resetting the TFT while executing prozesses on the Marlin board?

credomane, 2021-09-20:
> What is the status of the wrong Marlin configuration?

Ah sorry. Busy weekend and I never had time to get around to checking this. I went through and verified all the settings. Below is the exact checklist.

1. General options:
✔   EEPROM_SETTINGS (in Configuration.h)
✔   BABYSTEPPING (in Configuration_adv.h)
✔   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
✔   AUTO_REPORT_POSITION (in Configuration_adv.h)
✔   M115_GEOMETRY_REPORT (in Configuration_adv.h)
✔   M114_DETAIL (in Configuration_adv.h)
✔   REPORT_FAN_CHANGE (in Configuration_adv.h)

2. Options to support printing from onboard SD:
✔   SDSUPPORT (in Configuration.h)
✔   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
✔   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
✔   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

3. Options to support dialog with host:
✔   EMERGENCY_PARSER (in Configuration_adv.h)
✔   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
✔   HOST_ACTION_COMMANDS (in Configuration_adv.h)
✔   HOST_PROMPT_SUPPORT (in Configuration_adv.h)

4. Options to support M600 with host & (Un)Load menu:
✔   NOZZLE_PARK_FEATURE (in Configuration.h)
✔   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
✔   PARK_HEAD_ON_PAUSE (in Configuration_adv.h)
✔   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)

5. Options to fully support Bed Leveling menu:
❌   Z_MIN_PROBE_REPEATABILITY_TEST (in Configuration.h)
❌   G26_MESH_VALIDATION (in Configuration.h)
❌   Z_STEPPER_AUTO_ALIGN (in Configuration_adv.h)

>and resetting the TFT while executing prozesses on the Marlin board?

I don't understand...It behaves exactly how I described it in my original post.

oldman4U, 2021-09-20:
Well. I don't know too then. Sorry

kisslorand, 2021-11-19:
Since you reset the TFT and the problem persist (step 9, 10 in your list) it is obviously a Marlin issue.

oldman4U, 2021-11-19:
So let's wait another two months before this unrelated ticket gets closed.

👍🏻

JonathanHeinz, 2021-11-30:
Hello, i got the same problem. I think the problem is cause by the combined Marlin options EMERGENCY_PARSER and HOST_ACTION_COMMANDS.

closed 2022-02-05 by stale[bot]