bigtreetech/BIGTREETECH-TouchScreenFirmware #2182: [BUG] (In touch screen mode files on onboard SD card not visible)

DaddyCRO, 2021-09-17:
### Description

I am sorry if this ticket is already opened, but I was unable to find it.
Also, hope it is opened in right place.

Recently I updated my 3D printer with Marlin 2.0.9.1 firmware.
Before update, I was running Marlin 2.0.7.2 firmware (everything was working fine).

My HW configuration is:
* BIGTREETECH-SKR-E3-DIP-V1.1
* BIGTREETECH TFT24 V1.1 (BIGTREE_TFT24_V1.1.27.x firmware)

After Marlin firmware update, I cannot "see" *.gcode files on onboard SD card when they are in some folder (touch screen mode).
I see folder, can enter it, but *.gcode files are not visible in this folder.
If I put *.gcode files in a root of a SD card, files are visible.

If I switch TFT24 to Marlin mode, I "see" all folders and files: everything is as should be.

Since I upgrade main board to a Marlin 2.0.9.1 I also upgrade TFT24 firmware to version 1.1.27 (BIGTREE_TFT24_V1.1.27.x), but behavior is the same.

### Steps to reproduce

1. Upgrade main board to version 2.0.9.1
2. Upgrade TFT to 1.1.27
3. Create some folders on a SD card with some *.gcode files in it, also put some *.gcode files to a root of a SD card
4. put a SD card in main board SD card reader
5. switch TFT to touch screen mode
6. try to print from "OnboardSD"
7. *.gcode files in the root of the SD card should be visible
8. *.gcode files inside folders are not visible
9. switch TFT to Marlin mode
10. *.gcode files inside folders and in root are visible

**Expected behavior**
Folders structure with *.gcode files should be visible in both modes: Marlin and touch screen mode (this was the case until upgrade to a newer Marlin version).

**Actual behavior**
Explained above.

### Hardware Variant
BIGTREETECH-SKR-E3-DIP-V1.1
BIGTREETECH TFT24 V1.1

### TFT Firmware Version & Main Board Firmware details
Marlin 2.0.9.1 firmware
BIGTREE_TFT24_V1.1.27.x firmware

### Additional Information
I also tried with folder and *.gcode file names with less than 8 characters, but behavior is the same.

digant73, 2021-09-17:
try to update Marlin to last 2.0.9.1 (19 August).
It's working properly on my printer

radek8, 2021-09-17:
Only some folders and only some files in folders will be loaded on the TFT from the card on the board.
Isn't there a limit on the number of items that will be sent over the serial port on the TFT or Marlin side?
That would be logical. if there were hundreds of files on the card, the transfer would be inefficient.

DaddyCRO, 2021-09-17:
> try to update Marlin to last 2.0.9.1 (19 August).
> It's working properly on my printer

@digant73, I did update Marlin to 2.0.9.1 two days ago. And then it stops working...
At version 2.0.7.2 (I was running this version until upgrade), everything worked fine...

DaddyCRO, 2021-09-17:
> I have Malin bugfix-2.0.x from 25.8.2021
> Only some folders and only some files in folders will be loaded on the TFT from the card on the board.
> Isn't there a limit on the number of items that will be sent over the serial port on the TFT or Marlin side?
> That would be logical. If there were hundreds of files on the card, the transfer would be inefficient.

@radek8, I have SD card with only few folders and files. I did not change SD card content after upgrade.
Before upgrade, all files and folders were visible, after upgrade it is visible only folders in root. Subfolders are not visible.
Also, *.gcode files are visible from the root of SD card. If files are in subfolder(s), they are not visible.
And this happened only in "Touch screen mode". In Marlin mode everything works (and is visible) as before.

digant73, 2021-09-18:
My bad. Yes, the issue is present also with Marlin bugfix version

oldman4U, 2021-09-24:
Looks like a Marlin issue to me. They recently added multi storage support, could it be related?

What is the status of this ticket finally?

DaddyCRO, 2021-09-24:
Still not working for me.
But I am using Marlin mode to print parts.
Or touch screen mode and then print from USB stick.
But I prefer to print from a board SD card reader, so Marlin mode...

oldman4U, 2021-09-24:
But is it related to this firmware?

DaddyCRO, 2021-09-24:
Yes, 2.0.9.1 is not OK to me (regarding this bug). The previous version I had running was 2.0.7.2 and was OK.

oldman4U, 2021-09-24:
Yes, but this is MARLIN and not the TFT firmware. And this is the repository related to the BTT TFT firmware and not Marlin.

DaddyCRO, 2021-09-27:
I understand. I am sorry for not responding, but was out for a weekend...
Do you know where is Marlin related repository?
Just to check and if needed to open ticket there... Thank you...

oldman4U, 2021-09-27:
How could you update your Marlin without knowing where the repository is!?

https://github.com/MarlinFirmware/Marlin/issues

Please help the community and close this ticket once you do not need it anymore.

Thank you

PR #2284 (2021-12-14): Configurable common file extension suppression + bugfixes
**IMPROVEMENTS:**
* Configurable suppression (hide) for common gcode file extension: new param "filename_extension" added in config.ini. If set to 0 (false), any file extension starting with ".g" or ".G" (e.g. ".g", ".gco", ".gcode" etc...) will be hidden in both List Mode / Icon Mode menu. That setting can also be changed from Screen->UI menu. It implements #2074

**BUGFIXES:**
* Fixed missing display of filenames in sub folders when navigating from onboard SD on latest Marlin fw. It fixes #2182
* Minor bug fixes and optimization

reso [... cut]
closed 2021-09-27 by DaddyCRO