bigtreetech/BIGTREETECH-TouchScreenFirmware #2183: [Q]touch sensor is disabled.

Chicos123, 2021-09-17:
**Your question or details you need. Please describe.**
just updated V3.0.26 firmware to V3.0.27.x
The screen and the buttons work well, except for the touch.

**specify the device variant related to your question.**
device = BIGTREETECH-TFT35-V3.0
- firmware = BIGTREE_TFT35_V3.0.27.x.bin 
(_eed39913a55df4d5d97caa2f91c14a1bc323f5fd_ commit)
- THEME = THEME_Unified Menu Material theme\TFT35 
(_935482370c001cd05657b3eedb1c391b67f6a541_ commit)
- Language Packs = Language Packs
(_9243f26b8c39110a500942e23a44db8e09c320ec_ commit)

**Additional context**
How should I fix this? please help!

digant73, 2021-09-17:
try to recalibrate the screen. Put an empty file named "reset.txt" in the SD card and restart the TFT. Then follow the procedure reported on screen

Chicos123, 2021-09-17:
> 
> 
> try to recalibrate the screen. Put an empty file named "reset.txt" in the SD card and restart the TFT. Then follow the procedure reported on screen

It Worked! Thank You!

closed 2021-09-17 by Chicos123