bigtreetech/BIGTREETECH-TouchScreenFirmware #2184: [Q] (Sua Pergunta?)

ramosglauco, 2021-09-18:
Hi. I have an Artillery Genius (TFT28). I'm not able to show the layer number on the print screen, only the height is shown. Do I need to use some specific gcode between layers? PS: I'm using PrusaSlicer. Tks!

oldman4U, 2021-09-18:
Which version of the TFT firmware from this side are you using?

ramosglauco, 2021-09-18:
> Which version of the TFT firmware from this side are you using?

@digant73 version (1.27.x Patch 4)

oldman4U, 2021-09-19:
@digant73 

Does your repo include the code to show the layer?

digant73, 2021-09-19:
@oldman4U yes

oldman4U, 2021-09-19:
Ok. Can you reproduce the reported issue?

kisslorand, 2021-09-19:
Go to:
_PrusaSlicer -> Printer Settings -> Custom G-code -> After layer change G-code_
There insert the following:
`;LAYER:[layer_z]`

ramosglauco, 2021-09-19:
@oldman4U @kisslorand @digant73 
Got the layer number to be displayed on TFT. The correct code for PS is: 
; LAYER:[layer_num]
However two "-" are placed beside to the layer number if it is below 10. Example: "-3-". From the layer number 10 this does not happen. Is this a bug or is it correct?

kisslorand, 2021-09-20:
> However two "-" are placed beside to the layer number if it is below 10. Example: "-3-". From the layer number 10 this does not happen. Is this a bug or is it correct?

It's a bug, I will try to correct it ASAP.

ramosglauco, 2021-09-20:
> Also you can insert in PrusaSlicer at _Printer Settings -> Custom G-code -> Start G-code_ the following:
> `;LAYER_COUNT:[total_layer_count]` or `;LAYER COUNT:[total_layer_count]`

What are these commands in the start gcode supposed to show in the TFT? I tried here and nothing happened.

kisslorand, 2021-09-20:
With this comment you should also see the total number of layers of that gCode file.
You should see the layer number in format xxx/yyy, where xxx is the actual layer that is printed and yyy is the total layer number.

PR #2194 (2021-09-22): Fix - layer number display glitch
### Requirements

BTT or MKS TFT.

### Description

This little PR fixes a GUI glitch with Layer Number display during print. It occurred when the total layer number was unknown and only the printing layer number  was displayed. For the first 10 layers unwanted or extra characters were also displayed. This could also occur if the printing project contained multiple objects and the printing order was object by object and not layer by layer.

"Config.ini" is updated with info for PrusaSlicer users to have layer number info displayed on the TFT during print.

### [... cut]
closed 2021-09-22 by bigtreetech