bigtreetech/BIGTREETECH-TouchScreenFirmware #2188: No Support for an AZDelivery ESP8266

MoHa173, 2021-09-18:
### Discussed in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/discussions/2186

<div type='discussions-op-text'>

<sup>Originally posted by **MoHa173** September 18, 2021</sup>
I have congured an ESP8266 1MB from AZDelivery with ESP3D Software all sems to be fine wenn I have Connected to an UART to USB Converter I can connect via WIFI and configure but if i had connect to the ESP Port from BTT TFT35 V3.0 the ESP will not connect to my WIFI. I think the issue is that the ESP is always in Reset-state and the ESP8266 from BTT works in another way.

Can anyone explaine in wich way the ESP-Pins GPIO works and i can fix it with wiring or tell me the code ware the esp is initailized and i can change this.

Here the link to the Datasheet from the ESP wich i used:
https://cdn.shopify.com/s/files/1/1509/1638/files/ESP8266-01S_Modul_Datenblatt.pdf?100345

Thx for your help.

Kind regards

Moritz

ps: sorry for my bad english thats not my favorite language. Thats swabian :P</div>

oldman4U, 2021-09-18:
#113 includes everything related to your question

MoHa173, 2021-09-19:
> #113 includes everything related to your question

Big thx i have not found this tread i have soldered an 10k between VCC and EN now it work :)

is there any tread wich explaine how the auto shutdown works. I have customized an old Bequiet 80+ 300W Supply and will use the Autoshutdown on the PS_ON line of the supply.

oldman4U, 2021-09-19:
Usually the auto shutdown hardware comes with the information how to use.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-09-19 by MoHa173