bigtreetech/BIGTREETECH-TouchScreenFirmware #2190: [Q] (Can you enable the Marlin backlash setting with the TFT35 v3.0 screen?)

amv101, 2021-09-18:
I have a TFT35 v3.0 with the SKR v1.4 Turbo. I have enable the Configuration_adv.h [#define BACKLASH_COMPENSATION] setting in Marlin and it seems to work with the default Marlin screen mode (non-touch), but I cannot seem to figure out how to enable the setting to show on the touch screen interface. It seems like the setting works while in touchscreen, but there is no setting to adjust.

Is it possible to show this setting on the touch screen?

radek8, 2021-09-18:
Tato funkce není v dotykovém režimu implementována

amv101, 2021-09-18:
> Tato funkce není v dotykovém režimu implementována
Translated "This function is not implemented in touch mode"

Google to the rescue!

Would be nice to add this feature. After messing with it for a couple of days, backlash in Marlin seems to be a good feature. I had no idea it was even there. Not much of a pain to dial it in with normal screen mode for now.

radek8, 2021-09-18:
Sorry, transcribed in English :-)

amv101, 2021-09-18:
> Sorry, transcribed in English :-)

Haha no problem. So do you know if backlash even works with the touchscreen? Seems like it when I have it enabled in Marlin, the touch screen works sometimes then becomes unresponsive after a few button pushes. It wont even connect to the printer half the time, but when I switch over to the Marlin screen, it works fine. No issues in the Marlin screen at all with it enabled.

Is this a bug or just doesnt work at all in touchscreen mode?

radek8, 2021-09-20:
You can calibrate the clearance in Marlin.
If enabled in Marlin, of course, clearances will be used when printing from touch mode.

amv101, 2021-09-20:
> You can calibrate the clearance in Marlin.
> If enabled in Marlin, of course, clearances will be used when printing from touch mode.
> This is a function of Marlin

Hmmm it seems a bit buggy to me with the current setup. Maybe I need to update the firmware on the TFT35. As mentioned, it seems to work as needed in Marlin mode, but when I switch to touch mode, it works sometimes.... and then sometimes it doesnt. The printer becomes unresponsive in touch, it doesnt even show connected... then when I switch back to Marlin mode, it works fine.

radek8, 2021-09-20:
Always use the latest FW.
You write that touch mode sometimes works and sometimes it doesn't. This looks like a problem on the RS232 cable
Can you make a video that shows your problem?

oldman4U, 2021-09-24:
What is the status of the reported issue?

amv101, 2021-09-27:
Thanks for the help. Seems like after the firmware update to the screen and board, it doesnt seem buggy. Could have been the cable. Would still like backlash settings in the TFT menu somewhere.

closed 2021-09-27 by amv101