bigtreetech/BIGTREETECH-TouchScreenFirmware #2191: [Q] Marlin-Mode, Touch-Mode and UART to Raspberry Pi4 on 8bit Board possible?

tekknix, 2021-09-19:
Hi, 

i've converted my Sovol SV02 with the original 8bit Board (A Creality-like one), comppiled my Firmware with VSCode, got an BTT TFT7.0 and compiled the Firmware too and it works flawless.

I'm using Marlin-Mode and Touch-Mode, upload my Files via USB or SD-Card, make use of the Thumbnails from PrusaSlicer, ... i'm more or less very happy.

But, i'm missing a Remote-Connect Option to my Board/TFT.

The Sovol Silent Board offers EXP1, EXP2 and AUX1. On EXP1 & EXP2 the 10Pole Cables are used for Marlin-Mode, on AUX1 i've connected the RS232 Cable and now i can't use USB anymore if RS232 is active.

If i use Octoprint from my PI4, i'm not able to connect (okay, i can connect but nothing what i can do, No commands will be forwarded to the printer). As soon as i disconnect R232 everything is working fine.

I've tried to use UART3, UART4 or the Wifi-Connection on the BTT TFT and connected it to the correspondig TX and RX and GND on the PI4. 

But it isn't working at all. Thre is no connect. Tried it with Screen and with minicom. 

Are there any options left? Or only something like a SKR?

Any help will be appreciated.

Regards
MoS

oldman4U, 2021-09-19:
@tekknix 

How is your question related to the firmware of the TFT.

This is the repository related to the firmware and not a user group.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-09-19 by tekknix

radek8, 2021-09-24:
If you want to use UART 3 or UART4 on the TFT, you must first enable these ports on the TFT and set their communication speed.
By default, they are disabled to avoid interference.

radek8, 2021-09-24:
Another function has been added when the TFT switches to the mode when it does not communicate via RS232, but only listens to the communication of other devices and displays the read values. Maybe it's nude useful for you.

tekknix, 2021-09-24:
@radek8 : Many thanks for getting back. To be honest, it seems to be my Pi4/8GB. Even if do the same steps as on my Pi3B+ (where the same setup works).