bigtreetech/BIGTREETECH-TouchScreenFirmware #2193: [BUG] M150 Unknown Comand

Sekisback, 2021-09-19:
### Description

Added 3rd party Module OctoPrint-GpioRgbController to my octopi. This Module can control a normal RGB-Stripe, connected to RaspberryPins, via M150 command. I thought that i can control this stripe with the touchscreen from the LED Menu, because according to the code it uses M150 also.
I send the M150 R255 B255 U255 via the Octopi Terminal an I can control the LED Strip. I did the same on the Custom Command Buttons from Octodash, from here I can control the LED Strip too.

### Steps to reproduce

1. Pushing the Buttons in the LED Menu on / off red green blue, nothing happens
2. Try to send M150 R255 from the BTT TouchScreen Terminal, received the answer that M150 is an unknown Command

### Hardware Variant
BTT TFT35 V3

### TFT Firmware Version & Main Board Firmware details
TFT FW V.x.x.27
Marlin 2.0.9.1
OctoPrint 1.6.1 
Python 3.7.3 
OctoPi 0.17.0

guruathwal, 2021-09-19:
M150 from TFT only works for LED lights connected to the mainboard. You connected the RGB strip to your raspberry pi. You need to connect it to the mainboard.

Sekisback, 2021-09-20:
That is not the Point, the Point is that the M150 Command is unknown, independent of the connection

guruathwal, 2021-09-20:
That is the point. You will have to connect the LED strip to the mainboard and configure it in the marlin firmware first for it to work.

kisslorand, 2021-09-20:
> That is not the Point, the Point is that the M150 Command is unknown, independent of the connection

M150 is unknown for Marlin, so it is not a TFT FW issue.

Tannoo, 2021-09-21:
From the latest Marlin bugfix (Mine) `configuration.h` file:
```
// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB  // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  //#define NEOPIXEL_PIN       4   // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN      5
  #define NEOPIXEL_PIXELS     30   // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  //#define NEOPIXEL_IS_SEQUENTIAL // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST    // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

  // Use some of the NeoPixel LEDs for static (background) lighting
  //#define NEOPIXEL_BKGD_INDEX_FIRST  0              // Index of the first background LED
  //#define NEOPIXEL_BKGD_INDEX_LAST   5              // Index of the last background LED
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 }  // R, G, B, W
  //#define NEOPIXEL_BKGD_ALWAYS_ON                   // Keep the backlight on when other NeoPixels are off
#endif

```

Sekisback, 2021-09-22:
OK, got it... I had to learn, that the M150 from Octoprint works without changing the MarlinFW. case closed

closed 2021-09-22 by Sekisback