bigtreetech/BIGTREETECH-TouchScreenFirmware #2198: [Q] (Your Question?)

ramosglauco, 2021-09-24:
Hi devs! Is there any way to abbreviate the gcode filename the same way prusaslicer does the parts list? Exemple: Prefix...+...Suffix. This would help me a lot to better identify my files when printing, as the TFT (28) character limit causes the filenames to be cut off at the end. Thanks and sorry for my bad english...

[image]

oldman4U, 2021-09-25:
You never used DOS and the 8.3 filename limitation. 

😁

kisslorand, 2021-09-28:
@ramosglauco 
Isn't it just easier to manage yourself the file names upon saving them? Don't be that lazy! :)

closed 2021-12-06 by stale[bot]