bigtreetech/BIGTREETECH-TouchScreenFirmware #2200: [FR] UBL Grid probe

Chicos123, 2021-09-26:
**Is your feature request related to a problem? Please describe.**
UBL usually uses a 10x10 grid, so it takes a long time to probe when using the `G29 P1`.
Because of that i mainly use `G29 J5`. but TFT has no function to change `G29 P1`, `G29 J~`

**Describe the solution you'd like**
Plase add an option to switch`G29 P1`, `G29 J{1~GRID_MAX_POINTS}`commands
And an option to set the value of`{1~GRID_MAX_POINTS}`

oldman4U, 2021-09-27:
Whats about using a custom gCode for this?

Chicos123, 2021-09-27:
> Whats about using a custom gCode for this?

I usually set it up like this. 
```
G28    ; Home | XYZ
G29 L0 ; Bed Leveling | Load Mesh Slot | "0"
G29 P0 ; Bed Leveling | Phase 0 | Zero Mesh Data
G29 J5 ; Bed Leveling | Grid Probing | "5x5"
G29 P3 ; Bed Leveling | Phase 3 | Fill Unpopulated
G29 S0 ; Bed Leveling | Save Mesh Slot | "0"
G29 A  ; Bed Leveling | Activate UBL
```

oldman4U, 2021-09-27:
So 7 custom gCodes. Why not..

closed 2021-12-03 by stale[bot]