bigtreetech/BIGTREETECH-TouchScreenFirmware #2202: [BUG]  Message: TFT in listening Mode!

xrunner, 2021-09-27:
### Description

I am very familiar with updating the TFT firmware and have done it many times for over a year. The last two versions of BIGTREE_TFT35_V3.0_E3.27.x.bin show the following message at the top of the TFT screen -

**TFT in listening mode!**

I cannot use the TFT now to control the printer (but the TFT does work and I can go into it's menus), nor does it display any temperatures. I cannot find any mention of this in the configuration file or any other way to correct this issue in the config file or TFT sub-menus.

When I go into the Marlin mode I can then control the printer. Also, I can still operate the printer with Octoprint. But the issue remains, I cannot use the TFT touch screen to control the printer due to the new and never-before-seen message -

**TFT in listening mode!**

If I go back to a version of BIGTREE_TFT35_V3.0_E3.27.x.bin dated Sep 18, which I kept as a backup, this issue does not appear. I'm considering this a bug. Please advise.

### Steps to reproduce
Install the latest firmware BIGTREE_TFT35_V3.0_E3.27.x.bin, Unified menu for TFT35, and unaltered config.ini

**Expected behavior**
Expected the TFT to control the printer as usual and listen for temperatures.

**Actual behavior**
A message "TFT in listening mode!" appears at the top of the TFT screen in red and will not communicate with printer

### Hardware Variant
BTT TFT 35 E3 V3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.27.x.bin 

main board BTT SKR mini E3 1.2

### Additional Information
Configuration is exactly the same as the one provided in the .zip. If I go back to firmware dated Sep 18 the issue does not appear. I have never seen the message "TFT in listening mode!" before.

oldman4U, 2021-09-27:
check: Commits on Sep 22, 2021

oldman4U, 2021-09-27:
@digant73 Maybe this should be added to config.ini as well....

digant73, 2021-09-27:
if it was a configurable feature, it was already added in config.ini. But it is a feature like "disconnect" which you activate only from menu when you want it (e.g. temporary switch to offline).
@xrunner go to "Connection" menu and press on the new icon (on bottom left) to switch to "OFF". The listening mode will be deactivated. Did you provide the config.ini file installing the TFT fw? if so you should not have that issue

radek8, 2021-09-27:
You may need a more concisite icon label

radek8, 2021-09-27:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2179

xrunner, 2021-09-27:
> 
> @xrunner go to "Connection" menu and press on the new icon (on bottom left) to switch to "OFF". The listening mode will be deactivated. Did you provide the config.ini file installing the TFT fw? if so you should not have that issue

Great! That fixed it. I re-installed latest firmware and followed your instructions. Yes I'd have to say it's confusing and not obvious. Thanks and very impressed with the help in such a short time!

oldman4U, 2021-09-27:
Perfect.

Please close the ticket once you do not need it anymore.

Happy printing

closed 2021-09-27 by xrunner

digant73, 2021-09-28:
@radek8 I preferred to not define new labels/icons. Sure I know new labels could provide a better explanation but at the end pressing on the icon it will also display the "TFT in Listening Mode" text that I considered enough.
Unless I forgot to update the date (in the code), during the fw it should be asked to provide the config.ini file just to set Listening Mode to "disabled"