bigtreetech/BIGTREETECH-TouchScreenFirmware #2203: [BUG] Extruder calibration

radek8, 2021-09-27:
### Description

I noticed a problem in the TuneExtruder menu
if I calibrate the extruder, I can only set the new value of the remaining filament to the whole mm. For example, if I want to set 0.5mm, the value will change only one tenth, and more cannot be changed.

kisslorand, 2021-09-28:
I cannot reproduce the bug (MKS TFT28 v4).
There's something wrong with your FW. When you select adjustment step to 0.1 mm, the adjustments are done in a bigger step than 0.1 (as seen in the video on the link that you posted).

radek8, 2021-09-28:
@kisslorand
Are you saying you can't reproduce my mistake?
Then there must be a problem on my display, but I don't know what.
I restarted the display settings, loaded a new FW, a new config.ini, but the problem is still the same.

kisslorand, 2021-09-28:
Yes, what I say is that on my side the 0.1 step increases and decreases the value by 0.1mm (not more than that as seen on the video from the link you posted) and I can increase and decrease as much as I want by 0,1 mm.

radek8, 2021-09-29:
Thank you.
I think a change of 0.01 behaves the same as 0.1
I'm not at the printer right now. In the evening, I will try 0.01mm and also 10mm and I will answer your question.

digant73, 2021-09-29:
@radek8 eventually try to switch to english language and verify if the issue is still present

kisslorand, 2021-09-29:
I also had in mind that it could be a locale problem, that's why I asked him to try with 0.01 also. I saw that his TFT uses comma as decimal separator.

radek8, 2021-09-29:
Good
I will try all variants in the English environment.

radek8, 2021-09-29:
Yes, in the national environment, a comma is used as a decimal separator, a dot is used to separate thousands. 
For example: 1.158,35

radek8, 2021-09-29:
I tested the problem.
Yes, after switching to English the problem is solved.
I tried to set the Italian and Hungarian languages and the problem did not appear.
The problem is only if the setting is Czech language.
I rewrote the description 0.1 and 0.01 to numbers with a dot, but that's not the problem.
It is interesting that if I set a jump of 0.1mm and click on the button down, the value decreases by 0.12mm and 0.13mm, always on the inverter over and over again.
If I click up and I have a value set to less than 16.10 mm
the value increases once by 0.07mm and then 3x by 0.06mm and this is repeated up to a value of 16.1mm. Then it stops rising.

The problem is in the INI file with the translation ...

radek8, 2021-09-29:
I found a problem
The reason was line 301 in the translation, which read:
`label_tune_ext_old_estep:Původní E-kroky: %0.2f`

The problem was in the "ů" character.
I will go through the whole translation and remove this character to avoid further problems.

Thank you for your help and guidance in the right direction

closed 2021-10-06 by bigtreetech