bigtreetech/BIGTREETECH-TouchScreenFirmware #2205: [FR] Video or Webcam on Touchscreen

gordo3di, 2021-09-29:
**Describe the solution you'd like**
I'm sure I'm opening up a pandora's box here but I haven't seen any posts regarding implementing a webcam on the biqu touchscreens. I am currently using the TFT35 v3.

I researched and found people who can run video on the STM32 chips however I'm not sure how related it is.

https://www.youtube.com/watch?v=7-NHFz5MuoQ&t=248s

Is this feature even possible to implement?

**Describe alternatives you've considered**
Run video off a raspberry pi and separate touchscreen like octoprint etc...

oldman4U, 2021-09-29:
Great!

This could then replace my 75" 4K display in my home cinema;-)

gordo3di, 2021-09-29:
> Great!
> 
> This could then replace my 75" 4K display in my home cinema;-)

Haha. We could just plug the tv into the USB port with a splitter :D