bigtreetech/BIGTREETECH-TouchScreenFirmware #2207: Printing with udisk cold extrusion error

Charlestaylor216, 2021-10-02:
### Description

When I try to print with udisk it starts out fine but within a few minutes I get cold extrusion error and the print stops printing because it is to cold to extrude filament. I can print fine from SD card but not with udisk.

### Steps to reproduce

1. Print with udisk (Flash Drive)
2. [Second Step]
3. [and so on...]

**Expected behavior**
Pint and finish print.

**Actual behavior**
Stops printing and gives cold extrusion error.
### Hardware Variant
TFT35 E3 V3
### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

oldman4U, 2021-10-02:
Marlin and TFT FW version. A hint which makes you believe the fw is the reason for the problem?

radek8, 2021-10-03:
Same problem #706

oldman4U, 2021-10-06:
@Charlestaylor216 

It would be great if you would answer your own ticket.

Thank you

oldman4U, 2021-10-10:
@Msq001

Maybe you want to close this ticket. 

Thank you