bigtreetech/BIGTREETECH-TouchScreenFirmware #2208: [BUG] "Busy processing, please wait..."

yannick-alch, 2021-10-04:
Hi everyone,

### Description
I know this bug is known, but I start a new ticket.
When printing, sometimes I have a "Busy processing, please wait..." message ont the TFT 35 V2.0 screen, in red on top of the screen. When this message appear, I have to restart the printer (and then restart the print...).
I'm using TFT35 V2.0 screen,  [release Vx.x.26](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.26) and  SKR 1.4 motherboard.

### Steps to reproduce

1. Start many prints without restart or reset the printer.
2. Running at 115200 or 250000 baudrate

**Expected behavior**
Finish every print with success

**Actual behavior**
At a moment, you will see the message _"Busy processing, please wait..."_ on top of the screen
Then, the screen does not respond anymore at any command you want to send (G28, G1, M105, M114...)

### Hardware Variant
I am using TFT35 V2.0 screen,  [release Vx.x.26](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.26) and  SKR 1.4 motherboard with [marlin 2.0.5.3](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.5.3)
Also try with [release V.x.x.27](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/releases/tag/Vx.x.27) and same behavior

By searching on the source code of the TFT35, this is like the screen doesn't receive "ok\n" response from the motherboard, and then "infoHost.wait" is always "true" in the function "sendQueueCmd(void)" in interfaceCmd.c file. So the buffer is always full and wait till receive "ok\n" from SKR.
Maybe I am wrong, but this is my track.
I know @oldman4U , @Magiquall already work on it, but didn't works for me.

Hope somebody can help me deal with this...

oldman4U, 2021-10-04:
Sorry. I never saw this on one of my printer and I am not working on it.

Use the latest TFT FW and Marlin 2.0.9 and maybe someone is willing to help here.

Good luck

yannick-alch, 2021-10-04:
Not on your printer, but you already try to help someone else, in [this post](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/417) for example, so maybe you can have any suggestion for me.
Thank you

Magiquall, 2021-10-04:
Hello, 
personally I no longer have to worry at all with the latest firmware for the screen and marlin.  

This message still appears, but when I manipulate several commands at the same time, for example launching an automatic leveling then a home.  The message appears, but disappears as soon as the 1st instruction ends, normal operation as a result.

yannick-alch, 2021-10-04:
> Hello, personally I no longer have to worry at all with the latest firmware for the screen and marlin.
>
> This message still appears, but when I manipulate several commands at the same time, for [...]

How did you solve it in [this issue](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1837) ? Change firmware ?

Magiquall, 2021-10-04:
That's what I just told you, yes.  I know my English is bad, but still 😭

yannick-alch, 2021-10-07:
I made the change this morning, it is printing now

yannick-alch, 2021-10-07:
It also failed after 2hours of print...
I will try to reduce the baudrate to 115200

qrstu, 2021-10-08:
I have the same issues with BTT 1.4 SKR Turbo, I can't change as example "Flow" or "Speed" immediatly it could take up to 2-5min until it changes during printing. I'm also not able to "Cancel" a print on switch off is the solution. 
Printing works fine. 
Some times I get the message "busy processing, please wait"

My Marlin Parameters:
#define CONFIGURATION_H_VERSION 02000901
#define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
#define SERIAL_PORT 0 // MROE 2021-08-21 Fix TFT3
#define BAUDRATE 115200 // MROE 2020-05-15
#define SERIAL_PORT_2 3 // MROE 2021-08-21 Fix TFT35 --> Connected ESP3D onboard
#define BAUDRATE_2 115200   // MROE 2021-08-21 Enable to override BAUDRATE

BIGTREE_TFT35_V3.0_E3.27.x.bin --> 18-AUG-2021

Can anybody help ?
Thanks

yannick-alch, 2021-10-08:
This happend on about 15 TFT screen, so yes I believe it's a bug on the firmware

oldman4U, 2021-10-08:
How is it possible that the same firmware does work properly for others?

digant73, 2021-11-12:
@yannick-alch if the issue was fixed, please close this ticket

digant73, 2021-11-17:
@bigtreetech it seems the issue was solved by the user and no other help is required so the ticket could be close

closed 2021-11-17 by yannick-alch