bigtreetech/BIGTREETECH-TouchScreenFirmware #2210: [BUG] Marlin emulation mode not working

rubencm, 2021-10-04:
### Description

When I go to the Marlin emulation mode, it shows only a black screen with the text "Marlin mode"

When I connect a LCD12864 to the EXP1 and EXP2 pins, the LCD works correctly, but when I connect the cables to the TFT24 no.

### Hardware & Software
TFT24 GD with last version 1.1.27.x
Skr 1.4 turbo with DCDC module
Marlin version: 2.0.9.2 but wasn't working neither in 2.0.x-bugfix
My Marlin custom build [here](https://github.com/rubencm/Marlin/pull/2/files)

radek8, 2021-10-04:
What's your marlin record?

rubencm, 2021-10-04:
Skr 1.4 turbo

radek8, 2021-10-04:
Ok

radek8, 2021-10-04:
Can you take a picture of the wiring?

rubencm, 2021-10-04:
[image]
[image]

radek8, 2021-10-04:
The connection looks fine.
send the config.ini file

rubencm, 2021-10-04:
I think this is the config.ini i flashed the last time (before flashing it didn't work neither)

https://pastebin.com/WrpanmJp

radek8, 2021-10-04:
The parameters in the config ini are also fine.
if you used this configuration, it should work.
Try new uploads files, bin, font, icons and config.ini.

rubencm, 2021-10-04:
I aleady did that, but I will do again in a few minutes

rubencm, 2021-10-04:
I reflashed everything again (TFT24 folder, language_es.ini, config.ini and firmware file), and then did the reset.txt and still nothing

radek8, 2021-10-04:
If you disconnect the cable from the RS232 connector
Will the display remain powered on (will it remain lit)?

rubencm, 2021-10-04:
yes, and I can move the wheel and push buttons, I hear the click sounds, but I get no image

[image]

radek8, 2021-10-04:
Do you use the same cables you connect the TFT to connect the original 12864 display?

rubencm, 2021-10-04:
yes

radek8, 2021-10-04:
Since no one else who has the same display does not complain about the same problem, it will not be a FW TFT problem, but it looks like the HW problem of your display.

rubencm, 2021-10-04:
Thank you, I will try to contact the seller

closed 2021-10-04 by rubencm

wisetiger1701, 2022-09-29:
I'm having this exact issue with the same hardware. Did you resolve it?

rubencm, 2022-09-30:
No, I just got a refund

wisetiger1701, 2022-09-30:
Thanks for the quick response. Darn, I was really hoping there was a
software fix.