bigtreetech/BIGTREETECH-TouchScreenFirmware #2220: Marlin mode is not working!

muhammetbaygin, 2021-10-18:
I used mks sgen l v2 mainboard. and ı have tft24 v1.1 GD touchscreen. marlin 2.0.9.2 version.
[image]

avivarshney, 2021-10-19:
Configuration.h
https://pastebin.com/d7ZkpBZn

Configuration_adv.h
https://pastebin.com/TJUsQnj6

muhammetbaygin, 2021-10-19:
> Configuration.h https://pastebin.com/d7ZkpBZn
> 
> Configuration_adv.h https://pastebin.com/TJUsQnj6
> 
> config.ini https://pastebin.com/9KxeiExP
> 
> Image of Wiring https://imgur.com/a/luqr682

Sorry, links dont open

muhammetbaygin, 2021-10-19:
> @avivarshney in files configuration.h
> 
> comment // # define MKS_LCD12864B
> 
> uncomment #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Yes I did but not coming marlin screen

radek8, 2021-10-19:
@muhammetbaygin 
The advice was not for you.
You did not send the configuration.