bigtreetech/BIGTREETECH-TouchScreenFirmware #2231: I have the same issue with TFT50_v3.0 and TMC5160_v1.2.

magnificu, 2021-10-31:
I have the same issue with TFT50_v3.0 and TMC5160_v1.2.
When I start the print form the TFT SD Card, randomly the print freeze (sometimes after one or two hours of printing), but the hot end and the bed keeps the temperature. The only way to recover is to press the restart button.
I use the latest TFT firmware version BIGTREETECH-TouchScreenFirmware-master from 30.10.2021.

In the BIGTREETECH-TouchScreenFirmware official release-xx.27 this issue is not present.

_Originally posted by @magnificu in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1650#issuecomment-955647289_

oldman4U, 2021-10-31:
Okay. And you can tell us how this is related to the TFT firmware!?

magnificu, 2021-11-01:
With the old version of the firmware I don't have this issue or when I print from the SKR2 SD card.
This happens only with the latest firmware version and when I print from the TFT SD card.
Printing from octoprint is fine as well.
I think it is something on how the serial communication is handle. I tried 250000 and 115200 bound rates, but I had no luck.

oldman4U, 2021-11-01:
So please check the version which broke it by installing every version between the one which worked and the current one and let us know.

magnificu, 2021-11-01:
The one that is working is the release-xx.27 released on 10.04.2021. Since then I think there are hundreds of commits and I don't know how to test them. Is there any way to simulate the prints and to save all the serial communication on a log file?

oldman4U, 2021-11-01:
You download the version and test it. Maybe every tenth at the beginning until it breaks. I do not believe that there is a shortcut.

Keep in mind that you are currently the only one with this problem so besides the firmware version there is likely something else which causes the issue.

magnificu, 2021-11-01:
Is there any recommended setting for #define SDCARD_CONNECTION in Marlin 2.0.9.2 ?

Configuration_adv.h

  /**
   * Set this option to one of the following (or the board's defaults apply):
   *
   *           LCD - Use the SD drive in the external LCD controller.
   *       ONBOARD - Use the SD drive on the control board.
   *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
   *
   * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
   */
  #define SDCARD_CONNECTION LCD

radek8, 2021-11-02:
It depends on your preference.
LCD - SD card will be available on the display in Marlin mode.
In touch mode, only the SD card on the display and USB will be available. The SD card on the board will only be used to update MARLIN.

ONBOARD- in Marlin mode there will be an optional SD card on the board.
In touch mode, the SD card on the board + SD card on the display + USB will be available

Guilouz, 2021-11-05:
Very strange, I just have this issue too with printing with USB from TFT. Printer frozen, screen no longer reacts, reboot needed. Never had this problem before.

SKR2 with TFT70.

magnificu, 2021-11-05:
The question is how to tackle this?
The proposed solution to check all the commits for me sounds silly. I am surprised that there is no test/simulation environment.

radek8, 2021-11-05:
In the past, the problem of display blur when printing in touch mode was solved and was eliminated.
The return of the old problem?
You both have SKR2, but it will probably just be a coincidence.

oldman4U, 2021-11-05:
The problem has been solved by turning off all serial ports, so EMI does not harm the device. If someone with EMI turns serial ports on, he believes that there is a problem with the serial ports, which is not the case.

It would be silly to believe that this is a general firmware issue and that the device works for all the other users "by mistake". You wrote that an older firmware works well and does not show the issue (btw. "I have the same issue with TFT50_v3.0 and TMC5160_v1.2." is not an issue and you may want to change the title of this ticket), so checking which version broke it makes sense and even more, is maybe the only way to find out what broke it. Because you are the only one reporting this, it is up to you to find it and nobody else will be able to help you.

magnificu, 2021-11-05:
Thank you for your support and explanation.
I am not a developer myself so maybe I don't share the same view on how bugs are identified.
Can you please tell what is EMI?
Should I understand that if I download now again and compile it should work just fine?

oldman4U, 2021-11-05:
I am not a developer too.

https://en.wikipedia.org/wiki/Electromagnetic_interference

Because I do not believe that your problem is software related, I do not believe that downloading any version pre- or self compiled, will make a difference. But I would love to see you telling me which version works and which one breaks it, because then we would know which developer broke it, so he can fix it. But you think that this is silly, so I am not sure what you expect.

magnificu, 2021-11-05:
I just bought a product from BIGTREETECH and I want to use it without destroying hundreds of prints while looking for bugs.
release-xx.27 from 10.04.2021 works without any issues.
This is not the case with the latest github master, so there is a bug in the software.

oldman4U, 2021-11-05:
You can make test prints without filament loaded.

In your situation I would then use the version which works. If you are interested to find the version which broke it, you could load a newer version every time before you start a new print, until it breaks. Doing so you can use it without destroying hundreds of prints and the same time you will be able to find the version which broke it.

radek8, 2021-11-05:
May I ask if you use the display in English or if you use some localization?

magnificu, 2021-11-06:
The display is set in English and I written only the "language_en.ini"

magnificu, 2021-11-10:
In order to exclude the potential EMI issue, I want to provide additional information.
The control board and the TFT are position at ~450mm from the power source and at ~300mm from the closest stepper motor (z-axis).

radek8, 2021-11-10:
I have already met with the fact that the big source of EMI interference are stepper motor drivers ....
Try shielding the display with any metal foil attached to GND (the negative power supply).

digant73, 2021-11-10:
> Very strange, I just have this issue too with printing with USB from TFT. Printer frozen, screen no longer reacts, reboot needed. Never had this problem before.
> 
> SKR2 with TFT70.

Did you periodically update TFT fw and started to have issues only moving to SKR2 and TFT70? It can help to understand if the problem really started on TFT fw (and if so, the period the TFT fw broke) or possibly it depends on the HW

magnificu, 2021-11-10:
I don't think it is EMI since an older firmware version works just fine.

digant73, 2021-11-10:
@magnificu, @Guilouz can you provide a gcode file causing the freeze?

magnificu, 2021-11-10:
Yes, I can, is there any easy way to upload it?

digant73, 2021-11-10:
drag & drop here

digant73, 2021-11-10:
@magnificu flash the attached fw. Set the attached config.ini file according to you printer and flash it.
Let me know if you still continue to have freezes

[BIGTREE_TFT50_V3_0.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7516129/BIGTREE_TFT50_V3_0.zip)

magnificu, 2021-11-11:
Thank you for your support.
Here you can find the gcode attached. It is generated with prusaslicer and I could print it without any issues using an older firmware x.x.27 (release 10.04.2021)
I will test the firmware you propose and give you a feedback.

magnificu, 2021-11-11:
> @magnificu flash the attached fw. Set the attached config.ini file according to you printer and flash it. Let me know if you still continue to have freezes
> 
> [BIGTREE_TFT50_V3_0.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7516129/BIGTREE_TFT50_V3_0.zip)

Can you please tell me what did you changed in this firmware?
Typically I am compiling the firmware myself using Visual Studio Code.

digant73, 2021-11-12:
@magnificu I tested your file twice without any problem on last BTT fw released on 11/11/2021.
Since some months now, BTT fw need at least Marlin 2.0.8.1 and (as reported on top of config.ini file) that the following settings are enabled in Marlin:

# General options:
#   EEPROM_SETTINGS (in Configuration.h)
#   BABYSTEPPING (in Configuration_adv.h)
#   AUTO_REPORT_TEMPERATURES (in Configuration_adv.h)
#   AUTO_REPORT_POSITION (in Configuration_adv.h)
#   M115_GEOMETRY_REPORT (in Configuration_adv.h)
#   M114_DETAIL (in Configuration_adv.h)
#   REPORT_FAN_CHANGE (in Configuration_adv.h)
#
# Options to support printing from onboard SD:
#   SDSUPPORT (in Configuration.h)
#   LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
#   AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
#   SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)
#
# Options to support dialog with host:
#   EMERGENCY_PARSER (in Configuration_adv.h)
#   SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)
#   HOST_ACTION_COMMANDS (in Configuration_adv.h)
#   HOST_PROMPT_SUPPORT (in Configuration_adv.h)
#
# Options to support M600 with host & (Un)Load menu:
#   Options to support dialog with host (as pre requisite)
#   NOZZLE_PARK_FEATURE (in Configuration.h)
#   ADVANCED_PAUSE_FEATURE (in Configuration_adv.h)
#   PARK_HEAD_ON_PAUSE (in Configuration_adv.h)
#   FILAMENT_LOAD_UNLOAD_GCODES (in Configuration_adv.h)

I strongly suggest you to move to current (very stable) Marlin 2.0.9.2 applying the settings above and to use the latest BTT fw.

magnificu, 2021-11-12:
Thank you for your feedback.
I am using Marlin 2.0.9.2 and today I updated to the latest BTT fw version from 11.11.2021
I will check the options you proposed and test if the issue occurs again

digant73, 2021-11-12:
@magnificu in case you still continue to have freezes, try also to unplug the SD card and plug it again. Let me know if in this case the print is resumed (the freeze disappears)

digant73, 2021-11-12:
@bigtreetech as far as I see the function f_read used in loopPrintFromTFT does not give the control to the caller (loopPrintFromTFT) in case of errors (e.g. unplugging the SD card while printing) so the TFT freezes. Once you plug the SD card again, the TFT continues to properly work

Guilouz, 2021-11-17:
@digant just try your last PR and printing from TFT USB port, after 20h of print, printer stopped with "M1 Stop" message.

kisslorand, 2021-11-17:
@Guilouz
It happened to me the exact same thing, it was after 6 hours of print. It never happened to me before but I have a new MB, SKR2.
After this happened I changed the cable between MB and TFT to a shielded one and never happened again.

Guilouz, 2021-11-17:
@kisslorand what kind of cable do you use ?

kisslorand, 2021-11-17:
@Guilouz 
I used an USB cable with the shield soldered to GND (only at one end).

digant73, 2021-11-18:
@Guilouz my PR fixed a freeze printing from TFT SD. TFT USB was not affected by the bug. In any case (even if printing from TFT USB), if there is a read error during a print an error message should be now displayed and print should be aborted. In your case it seems more an EMI problem and as suggested by kisslorand you can probably fix it shielding the cable

Help3D-Padova, 2021-11-25:
> Also try to print from a SD or another USB stick

@digant73 same issue with Artillery X2 and the "new" Ruby 32 bit. Tried different Buffer size but M1 or M21 still appears random. As you said i think is related to the media device but the same Usb drive con the X1 is ok but in the X2 gives the M1 error. Same LCD FW installed on both

digant73, 2021-11-25:
I suspect it's more a bug on Marlin (for some boards) than an issue on the TFT fw. Since many months now, nothing has been changed in the handling of the print from TFT. I can suggest to try to disable from "Feature" menu some optional features such as Emulated M109/190, File Comment Parsing etc..., although I don't think they can cause any issue

magnificu, 2021-12-07:
I did not had this issue lately, so I assume it was due to EMI, even thought I did not changed anything on the hardware.
I will close this issue.

closed 2021-12-07 by magnificu

Guilouz, 2021-12-09:
Many people reported me they have M1 issue with SKR2 boards and also with Artillery Ruby boards on Sidewinder X2.