bigtreetech/BIGTREETECH-TouchScreenFirmware #2232: [BUG] (short description)

rubienr, 2021-11-05:
### Description

Discrepancy in between `Configuration.h` and `config.ini`.

### Steps to reproduce
Search for `neopixel_pixels` + `z_steppers_alignment` in `config.ini` and `NEOPIXEL_PIXELS` + `Z_STEPPERS_ALIGNMENT` in `Configuration.h`.

**Expected behavior**

| parameter | Configuration.h | config.ini |
|-|-|-|
| neopixel_pixels | yes |  yes |
| z_stepper_alignment | yes | yes |

**Actual behavior**

| parameter | Configuration.h | config.ini |
|-|-|-|
| neopixel_pixels | no |  yes |
| z_stepper_alignment | no | yes |

### Hardware Variant
All variants.

### Additional Information
That may be applicable for other arguments too. I did not verify them all.

radek8, 2021-11-05:
You can set some less frequently edited parameters only in configurstion.h
You can change most parameters in c config ini, without having to recompile the BIN file.
What's wrong with that?

rubienr, 2021-11-05:
Technically it is not wrong but that decision is interesting and cumbersome since everything that is in `.ini` must be parsed and depicted in code nonetheless. However, the point of truth shall be the `Configuration.h` whilst `config.ini` should be optional. 

Imagine following use-case: 
1. flash the display with `firmware.bin` + `config.ini` on CF card
2. remove the CF card
3. do a "Reset default settings"
4. now all settings that cannot be set in `Configuration.h` must be manually resurrected in Settigns -> Feature menu or by providing the `config.ini` again

Point 4. is feasible if the argument can be parametrized in Settigns -> Feature (for example: enable `z_steppers_alignment`), but what about `neopixel_pixels` which is not in the menue?

Off-topic:
To keep that consistent the least painful way would be to auto-generate the `.ini` file out of `Configuration.h` every time `#defines` are added/modified/removed.

oldman4U, 2021-11-06:
This is open source software. So please provide a PR which includes fixes for the things which you believe are a bug. 

Thank you

radek8, 2021-11-06:
You are right in your own way. But wouldn't the INI file be useless then?

FW was created to be easy to use for BFU as well. Just take the already generated BIN and then upload the INI file and the display is configured and ready. If something goes wrong or I change the printer configuration, just reload the INI file. This is at the cost of more flash memory consumption.
Unlike Marlin, where the emphasis is on the smallest possible FW size, but you always have to recompile the BIN file.
You wouldn't believe how many people eat there is a problem compiling the BIN file yourself.
If you have a problem, create a PR, as Oldman4U advises you.

radek8, 2021-11-06:
The fact that Configuration.h contains only some parameters and most of the configuration is in the INI file has a historical reason.
Originally, the configuration was only in Configuration.h, but as the number of parameters increased, the display would not be customizable without compiling.
Therefore, Cnfig.ini was created and the parameters that need to be set are transferred to it and it is further used for new variables.
Therefore, Configuration.h contains the original parameters and new parameters, which for some reason should not have been available in the INI file. Of course, if a new parameter was added to the INI file, it was no longer necessary to add it to Configuration.h.

Just to explain.

radek8, 2021-11-06:
[image]

commit 2ae334c1 (2021-11-10): Some improvements + bug fixes (#2233)

* resolves #2232
* setting inverted_ly_axis integrated into inverted_axis BIT value
closed 2021-11-10 by bigtreetech