bigtreetech/BIGTREETECH-TouchScreenFirmware #2236: Unknown Command. Missing spaces and chars on move commands

IrnScorpion, 2021-11-09:
Updated TFT35 V3.0 to TFT_V3.0.27.x Nov3 2021 on System Marlin 2.0.x (Github)
Printing from a microSD USB thumb drive. 

3 of 3 prints so far on this build i keep getting popups saying "Unknown command: G1X133.046 Y129.69 E2655.90902" or "unknown command: G X133.046 Y129.69 E2655.90902"
Characters or spaces are missing in random lines. 
The popup is on the screen but the printer continues printing. This happens multiple times on each print. 

After completing the print I confirmed the Gcode on the drive is correct.

Firmware issue? USB issue? microSD dying?

kisslorand, 2021-11-09:
- try printing directly from the TFT's SD card reader
 - try printing from a different SD card
 - try printing from an USB stick

radek8, 2021-11-09:
What speed do you use for serial communication? 
Are you using a squealed serial cable?

IrnScorpion, 2021-11-10:
Still running tests, but this is what i have observed so far. 

      > * try printing directly from the TFT's SD card reader
      > * try printing from a different SD card
      > * try printing from an USB stick

I took the microSD card out of the usb dongle and put it in a SD card adaptor, then installed that into the SD slot. I left the USB dongle (with no uSD card in it) in the USB port. Then printed the exact same untouched benchy file as before. Again I got unknown command errors just like before.
At that point in the print (~55% done), I removed the USB dongle and the print finished with no additional errors. 
On previous prints I had experienced multiple errors on one benchy print. 

I did notice that the USB dongle, even without a uSD card in it, will pop up the notification "u disk inserted". So having it installed is recognized by the system. 
So far this appears to be a continuation of the USB issues that earlier revisions had. 

This is not completely conclusive and I would still like to try USB thumb drives and different SD cards. Running more prints now.
I will continue to post updates as I discover.

      > What speed do you use for serial communication? Are you using a shielded serial cable?

Serial speed is 115200
No shielded cable. I am using the stock Ender3 ribbon cable.

IrnScorpion, 2021-11-10:
Tested 2 more uSD cards, same results. 
I also tested with and without the USB dongle, and it made no difference, still got errors.

digant73, 2021-11-13:
@IrnScorpion
1) If you are not on Marlin 2.0.9.2 yet, I suggest you to move to this version
2) In Configuration_adv.h (Marlin) use the following settings:
  \#define BUFSIZE 16
  \#define TX_BUFFER_SIZE 16

IrnScorpion, 2021-11-17:
Upgraded to 2.0.9.2. 
With just a few calibration prints so far, I have not see this issue yet. 

But 2.0.9.2 has given me new issues like
[BUG]Part Cooling Fan not Running - after clicking on Menu it starts with 100%, not adjustable - SKR Mini E3 V2 #20670

digant73, 2021-11-17:
ok, but this seems a wrong configuration on Marlin.

IrnScorpion, 2021-11-17:
Yes. I was just updating here so the ticket does not go stale and get closed. 
Just letting everyone know I want to follow up on this ticket, but cannot because some new issues need addressed first

digant73, 2021-11-17:
ok, but also remember to close the ticket as far as you fix the new problem (in case the original bug was fixed) ;-)

IrnScorpion, 2021-11-23:
Updated to the 2.0.9.2 released build and did not have this issue, but then my part cooling fan would not work.
That issue was resolved in the current bug fix.

closed 2021-11-23 by IrnScorpion

kisslorand, 2021-11-24:
> But 2.0.9.2 has given me new issues like [BUG]Part Cooling Fan not Running - after clicking on Menu it starts with 100%, not adjustable - SKR Mini E3 V2 #20670

Fixed: https://github.com/MarlinFirmware/Marlin/pull/23125