bigtreetech/BIGTREETECH-TouchScreenFirmware #2237: [BUG] (short description)

mikerocklewitz, 2021-11-10:
### Description
TFT70 latest firmware from github fails stating out of date icons.  They install fine.

### Steps to reproduce

1. Downloaded the repo from 11/9
2. Flashed tft70 with included themes (I tried unified material and hybrid red - both had same issue)

**Expected behavior**
Icon/theme to work

**Actual behavior**
Display says icons are out of date and need to be updated....even when updated

### Hardware Variant
TFT70

radek8, 2021-11-10:
Do you have icons on the SD card in the TFT70 folder, BMP subfolder?

mikerocklewitz, 2021-11-10:
> Do you have icons on the SD card in the TFT70 folder, BMP subfolder?

Yes of course.  I reverted to my download from 3 days ago and it works fine.

Simply reporting an issue with newest repo change.

oldman4U, 2021-11-10:
@digant73 

Can you see a possible reason why your PR causes the reported problem?

cooldude204, 2021-11-10:
the tft35 v3.0 also has this issue tried multiple icon packs and seems its trying to find 0.bmp and 2.bmp but can't find them.

digant73, 2021-11-10:
I also have a TFT35 v3 and no issue at all

cooldude204, 2021-11-10:
yes I'm using the precompiled version

digant73, 2021-11-10:
if you are familiar to compile the code, try to compile it. It's not currently possible for me to compile it for you

cooldude204, 2021-11-10:
I'm familiar with compiling marlin so I should be able to figure it out, I will compile my own .bin and let you know

digant73, 2021-11-10:
> I'm familiar with compiling marlin so I should be able to figure it out, I will compile my own .bin and let you know

You only need to set "default_envs = BIGTREE_TFT70_V3_0" in platformio.ini and compile

cooldude204, 2021-11-10:
Compiled the firmware uploaded it with icons on the SD card and its still getting the error icons: update required

mikerocklewitz, 2021-11-10:
I can try building it myself tomorrow but it looks like someone else found the issue as well.

digant73, 2021-11-10:
@cooldude204 

try if this works

[BIGTREE_TFT35_V3.0.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7516232/BIGTREE_TFT35_V3.0.27.x.zip)

cooldude204, 2021-11-10:
@digant73 unfortunately it didn't work. still says Icons: update required

cooldude204, 2021-11-10:
I changed branches to the release-Vx.x.27 branch and it works. and no failed 0.bmp and 2.bmp.

jeffeb3, 2021-11-11:
I'm not an expert at this codebase. So I hope I'm not causing problems here. But I suspect that one of the names of the icons has been chosen as a macro, and it is being resolved to 0 or 2 and then X_ICON ( 0 ) and X_ICON ( 2 ) is being called.

jeffeb3, 2021-11-11:
Aha!

LANGUAGE is defined in the Configuration.h and it is set to 0 by default
ONBOARD_SD is set in the Configuration.h and it is set to 2 by default.

PR incoming. I don't have a fresh screen to test on. Mine already have the 0.bmp and 2.bmp. How can I remove all of the icons already loaded, so I can test this?

bigtreetech, 2021-11-11:
fixed in #2239

closed 2021-11-11 by bigtreetech