bigtreetech/BIGTREETECH-TouchScreenFirmware #2246: I gave distance values ​​for each corner in L corner manual leveling, but it is not normal

newbie-0707, 2021-11-14:
The product is using the latest firmware on TFT35.

Currently using BL Touch.
With L corner, it moves relative to the BL touch probe. I wonder if this is correct. I want to manually set the L corner with the nozzle and A4 paper, but it doesn't seem to work.

And if the edge spacing is given by distance, the distance is correct for the edge close to the x/y limit s/w, but the distance does not match for the edge close to the x/y max value. I wonder if there is a way to set all of these to the same distance.

The printer is an anet A8 modified product and the board uses skr 1.4.

It was translated by Google, so I hope you can understand even if the sentence is strange.

radek8, 2021-11-14:
If you want to align the corners through the probe, click on Beb Level and L corner.
If you want to straighten through the probe and paper, click on Leveling.

[image]

digant73, 2021-11-17:
@newbie-0707
Yes, Level Corner menu uses the probe for the four edges. If you want to use the nozzle then you have to use Menu->Movement->Leveling menu.
For the wrong distances from the bed edges, it is a bug that will be fixed by #2141

radek8, 2021-11-17:
You probably have an older Marlin who has a bug.
The problem is the misreporting of the desktop dimensions by Merlin.
I reported the problem on August 19 and it has already been fixed.
The problem should not occur in the current build.

https://github.com/MarlinFirmware/Marlin/issues/22598

digant73, 2021-11-17:
@radek8 probably he has also an older Marlin version. In any case #2141 will fix the problem (even if work area is not provided by Marlin and the volume size on config.ini is used)

radek8, 2021-11-17:
#2241  :-)

Even if it writes the correct parameters to config.ini, the message from Marlin takes precedence and overwrites the value from config.ini. That's why I reported a problem in Marlin so it could be fixed.
Therefore, it is good to use the current version before I report a problem :-)

Yes, your fix will solve the problem for users who have a negative end stop and an outdated Marlin. Thank you

closed 2021-11-25 by bigtreetech

newbie-0707, 2021-11-25:
When I applied the latest firmware, the L corner distance was applied normally.

I want to say thank you to the person who replied