bigtreetech/BIGTREETECH-TouchScreenFirmware #2248: [FR] inverted (reverse) logic on steps (mm, percent, degree)

tispokes, 2021-11-14:
Every time i setup/do something manually, e.g. bed temp, nozzle temp, move x/y/z axis or leveling z-offet (and so on...) i would like to start with the biggest number first. Not with 1 Degree or 0.01 mm when i have to move first 1 mm then .4 mm and then .03 mm. so i have to press the icon very often.
It would be more useful to start from big and go smaller. (in my opinion, sure ;-))

PS:  Changing Degrees by 50 or 100 at first, would so nice. Think 50 is good to use for bed and nozzle. Betten than 10. And you want/need to go from 0 to 250 °C.

**Describe the solution you'd like**
config switch would be a nice solution. To setup directly in the settings would be great.

**Describe alternatives you've considered**
Writing a pull request by my own :-)

radek8, 2021-11-14:
Jumping to the lowest value cannot damage anything. An accidental jump by the highest value can damage something, such as the bed when the nozzle hits ...

tispokes, 2021-11-14:
Hmm okay, also on BabySteps it would not be nice. So excluding BabyStepping or possibility to setup each point itself.
Tried a bit, at the end it should not that hard.
Something like:
`uint8_t moveLenSteps_index = (infoSettings.inverted_steps * ITEM_FINE_MOVE_LEN_NUM) - infoSettings.inverted_steps; // index is 0 if disabled or max_index-1 (ITEM_FINE_MOVE_LEN_NUM) if enabled` per .c-file where it occurs
and when used to calc the new index:
`moveLenSteps_index = (moveLenSteps_index + infoSettings.inverted_steps + 1) % ITEM_FINE_MOVE_LEN_NUM;`

digant73, 2021-11-17:
@tispokes those kind of FR were already submitted and answered in the same way. As reported by Radek, it is always used the most safe solution (so the lowest values). Also, once changed from a menu, the values will maintained in case you re-enter in the menu. So at the end, a minimal effort is requested to the user

tispokes, 2021-11-17:
Hmm no the effort stays the same. e.g. ich recalibrate my z-offset:
1) 1 mm step (pressing mm-rotation-button 2 or 3 times)
2) 0.1 mm step (pressing 2 oder 3 times again)
3) 0.01 mm steps (and again 2 or 3 times)
Then im back at 0.01 mm and then, when i change the bed and want to recalibrate again i have to do the same steps again

digant73, 2021-11-17:
I meant the changes (the last values you select) are stored until you restart the printer/TFT

tispokes, 2021-11-17:
Yes therefore i end the callibtation with 0.01 and as next i get 0.1 again. (updated commend)
So reverse order would be great. may i have to write and test ir myself with optIN/OUT.
Also to secure inexperienced user, the cycle could start at 0.01 and the next is 10/1 and then lowering from there.

tispokes, 2021-11-17:
And i restart the printer more often then i change my z-offset ;-)

closed 2022-01-24 by stale[bot]