bigtreetech/BIGTREETECH-TouchScreenFirmware #2255: [BUG] TFT no more able to connect to Marlin

digant73, 2021-11-18:
Just to avoid panic, last merged PR #2250 made the TFT not able to connect to mainboard at startup for any fw different than RepRap.
The issue is fixed in PR #

oldman4U, 2021-11-18:
I always wanted to write you this once. 

Please help the community and close the ticket once you do not need it anymore. Thank you

🤗

pfn, 2021-11-22:
That's interesting, checksums should still work on non-RRF firmwares. I wonder why not... maybe it's the missing line-number reset. I guess this will have to be a specific config.ini option, sadly  :(

xPew, 2021-11-22:
> That's interesting, checksums should still work on non-RRF firmwares. I wonder why not... maybe it's the missing line-number reset. I guess this will have to be a specific config.ini option, sadly :(

Well, for now there is nothing specific in rrf config.ini so there is place for it… :D

digant73, 2021-11-23:
@pfn the problem is that the TFT initial state is FW_NOT_DETECTED. Only after connection is established, the state is changed. But in case of non RFF, no message could be sent to printer

pfn, 2021-11-23:
It was intentional as the expectation was that marlin and other firmwares understand checksums (it's a standard). Thanks for fixing it, will need to add a config option for sending checksums even when the firmware is not yet detected.

r00t4rd3d, 2021-11-23:
I just flashed the latest firmware for the TFT35 v3.0 and now my printer wont start. Says stuff about Liner number not being last line number. I had to revert to firmware from a couple months ago to fix it. 

[image]

aldaw, 2021-11-24:
> I just flashed the latest firmware for the TFT35 v3.0 and now my printer wont start. Says stuff about Liner number not being last line number. I had to revert to firmware from a couple months ago to fix it.
> 
> [image]

Dito, i have the same problem, i thought i was because of marlin but after 3 diffrent .x versions i guess it is the firmware for the screen.

pfn, 2021-11-24:
Yep, this is my bad, should have set the first line number to be sent as 1, instead it's 0, and that causes this error.

pfn, 2021-11-24:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2250/files#diff-edea03df047c28a4f27608dbf5360fd4ab229b4cf35667fc89761e74b37f8980R6 this could be fixed by initializing line_number to 0, or pre-incrementing with ++line_number

aldaw, 2021-11-24:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2250/files#diff-edea03df047c28a4f27608dbf5360fd4ab229b4cf35667fc89761e74b37f8980R6 this could be fixed by initializing line_number to 0, or pre-incrementing with ++line_number

So I would have to compile it i guess that i can test it?
Or could I try to use this command?
https://marlinfw.org/docs/gcode/M110.html
Example how i guess it:
M110 N0

pfn, 2021-11-24:
@digant73 already "fixed" this by removing checksum lines for the not yet detected firmware case.

And it'd have to be M110 N-1 if that were possible.

r00t4rd3d, 2021-11-24:
> @r00t4rd3d to wich version did you fall back? Where can i find it?

Go here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Then click here:

[image]

Then click here:
[image]

This will bring you back to that release date.

aldaw, 2021-11-24:
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2250/files#diff-edea03df047c28a4f27608dbf5360fd4ab229b4cf35667fc89761e74b37f8980R6 this could be fixed by initializing line_number to 0, or pre-incrementing with ++line_number

I tried now also pre-incrementing with ++line_number instead of line_number.
Either I am to stupid or some thing else is missing :P

aldaw, 2021-11-24:
> > @r00t4rd3d to wich version did you fall back? Where can i find it?
> 
> **Go here:**
> 
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
> 
> **Then click here:**
> 
> [image]
> 
> **Then click here:**
> 
> [image]
> 
> This will bring you back to that release date. I can not remember the exact build I used, I just went back a couple months and used a Verified release. My printer is running right now so I cant check and see what version I am using.

Thank you so endless much a lot <3 was looking for version histories on github for a long time!
Will try it after i got some sleep, tried to fix that for 5 hours xD

pfn, 2021-11-24:
@aldaw 
> I tried now also pre-incrementing with ++line_number instead of line_number. Either I am to stupid or some thing else is missing :P

it's kind of an annoyance, but marlin numbers will march on, so if the number gets out of sync, this error happens (reboot marlin w/o tft, or vice versa). it could probably be a more comprehensive fix with checksums always being sent by reading that error message, and updating the line number for future commands. for now, it's easier to just disable sending checksums if it's not RRF (RRF doesn't care about the line number really)

aldaw, 2021-11-25:
For anyone who encounters the same bug. i used like @r00t4rd3d said a older version.
To be specific from Nov 11, 2021 and now i works like a charm nothing annoying! :)
Here a link for faster use https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/78c41d021ad0cb1330eabaa8c90e6c95b6824dd4

closed 2021-11-25 by bigtreetech