bigtreetech/BIGTREETECH-TouchScreenFirmware #2258: [BUG] (short description)

MLiv79, 2021-11-19:
### Description

Print killed for Heat
### Steps to reproduce

When i print ABS i use to heat my bed to 110°. If i print from tft around 100/103° the print fails for heating reasons (pint killed ....) while if i print from marlin with the same gcode, just switching the sdcard to my mainboard, the print starts and finish without issue....
What could be the cause of this difference?

**Expected behavior**
Print at 110° bed temp both from marlin and from tft
**Actual behavior**
From TFT it doesn't work
### Hardware Variant
TFT 70
### TFT Firmware Version & Main Board Firmware details
Release of 17 november TFT + Marlin 2.0.9.1
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

kisslorand, 2021-11-19:
> If i print from tft around 100/103° the print fails for heating reasons (pint killed ....)

What do you mean? Temperature drops to 100/103° from 110° or when heating up and reaches 100/103° ?

MLiv79, 2021-11-19:
> > If i print from tft around 100/103° the print fails for heating reasons (pint killed ....)
> 
> What do you mean? Temperature drops to 100/103° from 110° or when heating up and reaches 100/103° ?

When the printer heat up, it stops at around 100° for "print killed heat...." i don't remember the exact message. Seems that, because of it took time to reach 110°, the TFT has some setting that kill the heating process after some time or after that temperature doesn't change for X seconds. In marlin mode it doesn't happen.

gordo3di, 2021-11-19:
What's the max temp set to in your TFT config file?

MLiv79, 2021-11-19:
> What's the max temp set to in your TFT config file?

150

digant73, 2021-11-19:
Try to disable Emulated M109/M190 feature on TFT (on config.ini, set "emulated_m109_m190" to "0" or disable Emulated M109/M190 directly from Feature menu).
If it doesn't work, try to increase WATCH_BED_TEMP_PERIOD in Configuration_adv.h in Marlin.

MLiv79, 2021-11-19:
> Try to disable Emulated M109/M190 feature on TFT (on config.ini, set "emulated_m109_m190" to "0" or disable Emulated M109/M190 directly from Feature menu). If it doesn't work, try to perform PID with 120° for the bed and/or increase WATCH_BED_TEMP_PERIOD in Configuration_adv.h in Marlin.

I'll try to disable M109-190 emulation first. Pid is recently performed (65° as always) and watch bed temp period in marlin should be correct because if i print from marlin, instead of from TFT, it works correctly

MLiv79, 2021-11-19:
> Try to disable Emulated M109/M190 feature on TFT (on config.ini, set "emulated_m109_m190" to "0" or disable Emulated M109/M190 directly from Feature menu). If it doesn't work, try to perform PID with 120° for the bed and/or increase WATCH_BED_TEMP_PERIOD in Configuration_adv.h in Marlin.

Thanks @digant73, disabling M109-190 emulation worked...

digant73, 2021-11-20:
@MLiv79 good. Remember to close the ticket if you don't need support anymore

closed 2021-11-20 by MLiv79