bigtreetech/BIGTREETECH-TouchScreenFirmware #2263: [BUG] (short description)

NOVAXIM, 2021-11-24:
### Description
The problem refers to switching from Marlin mode to touch mode.

### Steps to reproduce

1. If printer is touch mode I set Marlin mode.
2. I back to touch mode.
3. Printers beeps intermittently and displays it:
[image]
Note: First time displaying '7' instead '5'.

**Expected behavior**
Please write if your TFT works the same (don't forget to mention information about hardware, firmware, and configuration). I am asking for any information that will help me to a quick solution.

**Actual behavior**
When I use "BIGTREE_TFT35_V3.0.27.x.bin" dated 25 Oct then switching work well (with old images: "marlin.bmp", "bigtreetech.bmp").

### Hardware Variant
SKR-mini-MZ V3.0 with TFT35-MZ V3.0

### TFT Firmware Version & Main Board Firmware details
Problem was found on the last compilation of November 21st and earlier (with new images: "Marlin_mode.bmp", "touch_mode.bmp").

### Additional Information
Refers to the standard configuration.

aldaw, 2021-11-25:
We had the same problem solved it with using a older version.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2255

bigtreetech, 2021-11-25:
fixed in #2241. Please confirm, Thank you very much!

closed 2021-11-25 by bigtreetech

NOVAXIM, 2021-11-25:
Yes, it is fixed. Thank you!