bigtreetech/BIGTREETECH-TouchScreenFirmware #2265: Printer not connected - Line Number is not Last Line Number+1, Last Line: 0

NestorUnzu, 2021-11-25:
**Your question or details you need. Please describe.**
I'm trying to set up a TFT35 v3.0 with a Bricolemon Lite Board. 
Marlin-2.0.9.1 and TFT FW version BIGTREE_TFT35-v3.0.27.x
When I boot the boards I get the "No printer connected" in the top of the TFT followed by a "Line Number is not Last Line Number+1, Last Line: 0" pop up and constant buzzer beeping. 

If I insert an SDCard with the "reset.txt" blank file and reset the boards, after calibrating the TFT the system works fine but as soon as I reboot the boards I get back to the above errors.

If I load the TFT FW version BIGTREE_TFT35-v3.0.26 everything works fine no matter how many times I reboot the boards.

**specify the device variant related to your question.**
TFT 35 v3.0

**Additional context**
I could use the FW 26 version but I need the TFT menus to be in Spanish and I can't, for the life of me, find a full release of the FW v26 to be able to configure the langage.

kisslorand, 2021-11-25:
Use latest version, it's been fixed.

NestorUnzu, 2021-11-25:
I understand that you mean that I should use the latest version of the TFT FW, I'm seeing there's a release from 5h ago. The FW giving me the problems was dowloaded yesterday.

Going to test this new release and I'll get back to you.

Regards,

NestorUnzu, 2021-11-25:
Hi again,

I can confirm that the issue is indeed solved in the new FW release.

Thank you @kisslorand for your prompt reply.

closed 2021-11-25 by NestorUnzu