bigtreetech/BIGTREETECH-TouchScreenFirmware #2268: [BUG] Text flipped after firmware update

TeriyakiBomb, 2021-11-25:
### Description

I tried to update some graphics with the newest firmware, afterwards, the text on the screen is flipped horizontally, also noteworthy is my initial firmware had a red logo and was dated June 2021. The latest firmware here is from April, so what's that about? I've also tried re-downloading a fresh zip, trying the previous firmware and nothing is working. I'm using the ender 3 version of the screen and I have been selecting the correct .bin, I've tried swapping themes, looked through the config file, reformatting my SD card (2gb). Basically everything I can think of.

At least one other person has had this issue and resolved it, but did not provide the solution on reddit https://www.reddit.com/r/BIGTREETECH/comments/r0s4ta/btt_tft35_e3_v3_flips_orientation_when_updating/

### Steps to reproduce

Follow official instructions

**Expected behavior**
Updated firmware

**Actual behavior**
Screen is initially rotated 180 degrees, text is also written in reverse (The reddit link shows this) which is still happening once the screen is rotated correctly

### Hardware Variant
TFT35 E3

### TFT Firmware Version & Main Board Firmware details
Main board is a skr trubo E3, using BTT firmware (unmodified)
TFT firmware was initially .27, but a build dated June 2021, rather than April. It had a red boot logo instead of the blue from the .27 found on github

bigtreetech, 2021-11-26:
Hello, have tested this .bin https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT35_V3.0_E3.27.x.bin ?

TeriyakiBomb, 2021-11-26:
Yeah, this worked. Thanks. Will this be added to a formal release soon to save all of the hunting?

radek8, 2021-11-26:
> Yeah, this worked. Thanks. Will this be added to a formal release soon to save all of the hunting?

this is the official release

TeriyakiBomb, 2021-11-27:
It isn't. It's what's on master, not Vx.x.27 from April. 

My point being, if bugfixes are being made, you guys should maybe look at doing more than one release a year.

radek8, 2021-11-27:
Master is the only release you have to use

radek8, 2021-11-28:
Please close the ticket if your issue is resolved.
Thank you

closed 2021-11-30 by bigtreetech

diabeetus01, 2022-02-21:
This is happening for me now, every single firmware available from you guys for TFT35 has flipped text exactly like the photos on the OP's linked Reddit post. I was just trying to update the firmware and now I can't use the screen, as every .bin I try out has this exact same issue.

kisslorand, 2022-02-21:
Have you tried the solution given? https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2268#issuecomment-979646133

radek8, 2022-02-21:
read here
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2411

diabeetus01, 2022-02-21:
Yes, and have the same result, as that is the file I originally flashed my updated firmware with.
It looks like it is mirroring the text, or drawing it right-to-left.
[image]
[image]

Now flashing the firmware in the thread linked by radek8, so far looks promising, the updating screen is showing text in the proper direction this time.

diabeetus01, 2022-02-21:
Update: SUCCESS with the linked solution, thank you, wasn't able to find that trying to search around.