bigtreetech/BIGTREETECH-TouchScreenFirmware #2272: [FR] Minor UI changes

xPew, 2021-11-28:
To better suit my theme, I would like to ask for some minor UI changes but would be a nice touch.
Here, I would like to change the status screen text color to white. Can it be added as an option to the config.ini? My theme has a color scheme due to how our web interface looks, so I don't want to change it to a lighter background if it's possible to change the text color.
I guess nobody wants to use fancy colors here, so only a black and white option is enough, not to consume a lot of memory.
[image]

My other request is to @kisslorand . I really like your PR's, especially the status bar you added back then. May I ask you about the possibilities of its customization? I am not a programmer, and I guess it all comes down to memory issues, but I would like to also tweak this part of the UI to better suit my theme.  Could some color choices be added for this part of the UI too? Better in config.ini, but if it's not possible maybe some guidance about what to change in case of self-compiling to be able to customize this.

Thank you in advance! :)

radek8, 2021-11-28:
All you need is an adjustment

[image]

kisslorand, 2021-11-28:
> My other request is to @kisslorand . I really like your PR's, especially the status bar you added back then. May I ask you about the possibilities of its customization? I am not a programmer, and I [...]

Thank you for your appreciation but the status bar/icon is not my work. Anyhow What customization do you have in mind? What color choices and where do you have in mind?

xPew, 2021-11-28:
> Thank you for your appreciation but the status bar/icon is not my work. Anyhow What customization do you have in mind? What color choices and where do you have in mind?

Oh, sorry I ment the progress bar during printing. I would like to be able to modify the color of if. In my case blue, but I guess if a few basic color would be added others could benefit from it too.

kisslorand, 2021-11-28:
OK, I got it now.
I will try to come up with something to be able to select between variants and/or customize the progress bar color.

radek8, 2021-11-28:
@kisslorand 
it is not a matter of changing the color of the progress bar but of changing the color of the digits (temperature of the nozle, bed, fan ...) on the status screen.
See image above.

radek8, 2021-11-28:
The current black is not visible on a gray background.
@xPew  Can't you make the gray one lighter so that the black font is visible?

kisslorand, 2021-11-28:
> @kisslorand it is not a matter of changing the color of the progress bar but of changing the color of the digits (temperature of the nozle, bed, fan ...) on the status screen. See image above.

I beg to differ, see the comment of @xPew above.

"Oh, sorry I ment the progress bar during printing. I would like to be able to modify the color of if. In my case blue, but I guess if a few basic color would be added others could benefit from it too."

radek8, 2021-11-28:
Personally, I see no reason to use a gray theme on a color display.
I like colors :-)

radek8, 2021-11-28:
@kisslorand  Probably lost in translation

xPew, 2021-11-28:
@kisslorand and @radek8 I requested both thing. The status screen font color and the progress bar color.
Thank you kissroland if you could see what you can do.
Also, radek, I copied the theme from our web interface (Duet Web Control), that's why it's so monochromatic. While I partly agree that it could be more colourful but I imagined this theme to be as close as possible, so I would like to ask for white fonts in the first place.

radek8, 2021-11-28:
@xPew  did you try the edit I posted in my first comment? You will have white digits

xPew, 2021-11-28:
> @xPew did you try the edit I posted in my first comment? You will have white digits

I need to copy the exact line? I am not a programmer and I don't see and white in that.
If yes I will try it, but I would like a more permanent/comfortable solution.
Thankfully this firmware is updating daily, so modifing it every time is a bit of a hassle.

kisslorand, 2021-11-28:
Well... Doing firmware changes on requests is also a hassle.... :)

radek8, 2021-11-28:
> > @xPew did you try the edit I posted in my first comment? You will have white digits
> 
> I need to copy the exact line? I am not a programmer and I don't see any white in that. :D If yes I will try it, but I would like a more permanent/comfortable solution. Thankfully this firmware is updating daily, so modifing it every time is a bit of a hassle.

Yes
the red line is the original and is replaced by the green one.
Then you just compile. and will be white.

radek8, 2021-11-28:
Use GutHub Desktop.
Working with new versions and custom adjustments is then very simple ...

kisslorand, 2021-11-28:
I kind of finished adding Orange, Yellow, Red, Green, Blue, Gray colors to the progress bar but it is too late here (over midnight) so I will test it tomorrow and make changes if needed.
It is not configurable in "config.ini", it will be rather configurable at compile time ("Configuration.h"). This way it adds 0 (zero) to the compiled program size.

radek8, 2021-11-28:
Likewise, the white numbers in the live icons could be added during compilation

radek8, 2021-11-28:
Purple Nooo

kisslorand, 2021-11-28:
> Purple Nooo

Maybe Pink? LOL

radek8, 2021-11-28:
Good night

xPew, 2021-11-28:
Thank you guys! Gn!

xPew, 2021-11-29:
Is there any reason why the status screen fonts are fixed black? In config.g ## Menu font color - menu_font_color:0 can be customised, but it dont change the status screen fonts. Can it be combined to prevent workarounds, and also make this setting more sense? This was my first attempt when I wanted to change the color and dont understood why it stayed black…

thisiskeithb, 2021-11-29:
I’d like to see a “standard” set of colors like what’s used elsewhere in code:

- WHITE
- BLACK
- RED
- ORANGE
- YELLOW
- GREEN
- BLUE
- INDIGO
- VIOLET
 
Users will be familiar with those options and can match their UI colors to whatever LEDs or icon scheme they desire.

kisslorand, 2021-11-29:
I do not see anywhere INDOGO and VIOLET in the code for TFT pixel colors.. Maybe it is just me but I have hard times imagine a BLACK progress bar.

kisslorand, 2021-11-29:
@xPew 
Check PR #2272.

bttguy, 2021-11-29:
Wow, nice! Check it when I next to my printer. Out of curiosity how much memory would it take if a config.ini was used to set this?

kisslorand, 2021-11-29:
> Wow, nice! Check it when I next to my printer. Out of curiosity how much memory would it take if a config.ini was used to set this?

I don't know until I make it that way.

xPew, 2021-11-30:
> @xPew Check PR #2272.

Seems fine to me. Its blue now :)

xPew, 2021-12-01:
I saw the PR is merged. Thank you!
May I ask for a config.ini implementation of both the status screen font color and the progress bar color? At least to know wether it can fit or not.
Not urgent, but I would like to keep this FR open a bit more :)

radek8, 2021-12-01:
check #2276

PR #2276 (2021-12-04): Optional value change on the status screen
Added the ability to change the color of the values on the live icons on the status screen in the configuration.h file.
Some topics require the inverse display of color values on live status screen icons.
For example: THEME_Rep Rap Firmware Dark
resolves #2272
closed 2021-12-04 by bigtreetech