bigtreetech/BIGTREETECH-TouchScreenFirmware #2274: [BUG] (short description)

NOVAXIM, 2021-11-29:
### Description
Such strange anomalies occur when printing in Marlin mode. The longer I print, the more errors on the screen. The photo shows the condition after approx. 3 hours of printing.
[image]

### Hardware Variant
SKR-mini-MZ V3.0 with TFT35-MZ V3.0

### TFT Firmware Version & Main Board Firmware details
It concern to prebuild 80ae8d6d917563aae2986efc3a53e7998e0cf575.

### Additional Information
In addition, I have noticed that the printer sometimes stutters when printing in Marlin mode. WHen used earlier prebuilds c9b326ced55f63bb911364edfb3716ec3240f7e1 I did not notice it.

radek8, 2021-11-29:
If you click on the Knob to open the menu and return to the status screen. Will the display be corrected?

radek8, 2021-11-29:
The inscription Marlin mod is fine. It looks like the data is coming to the display already corrupted.
How do you have long EXP cables?

NOVAXIM, 2021-11-29:
> If you click on the Knob to open the menu and return to the status screen. Will the display be corrected?

I forgot to mention about it. Of course, I tried it and nothing change. After it finished printing, only resetting the printer fixed the screen.

> The inscription Marlin mod is fine. It looks like the data is coming to the display already corrupted. How do you have long EXP cables?

You think right, but I use the original BigTreeTech cable from set, which is for my printer designed. Maybe WiFi module?! But why does the printer stutter? After rebooting and starting the same Gcode, the printer did not stutter while printing. These stuttering bothers me more.

Currently, I have a lot to print and will be watching closely. I will describe the observations. Fortunately, the prints are beautiful.

NOVAXIM, 2021-11-29:
About cables... you mean about black (TFT) or grey (EXP)?

radek8, 2021-11-29:
I asked about the EXP cable

The black 5-pin cable (RS232) uses touch mode for serial communication.
The gray 10-pin EXP1 (or EXP3) cable uses Marlin mode to display the Marlin screen. EXP2 uses Marlin to access the SD Card.
If you print from Marlin. Is the SD card in the Marlin board or in the TFT display slot?

NOVAXIM, 2021-12-01:
Zgadza się. I used board microSD because my ANYCUBIC access to TFT sockets (USB and SD) is difficult (by housing). That is why I am very surprised by stuttering in Marlin mode, while in touch mode everything is ok.

radek8, 2021-12-01:
This looks like a problem on your SKR mini MZ board.
If you are printing from the onboard SD card slot, the TFT display cannot affect stuttering in the print.
Do you have another display with the EXP3 connector to test how the image on another display will behave?

digant73, 2021-12-04:
in case it's not caused by the usual EMI, the problem can be related to the delays set in Marlin.

radek8, 2021-12-04:
@NOVAXIM, 
can you attach a Marlin configuration?

NOVAXIM, 2021-12-05:
@radek8, you mean about Configuration.h or Configuration_adv.h?

radek8, 2021-12-05:
both

oldman4U, 2021-12-09:
What is the status of this ticket?

Snargaloo, 2021-12-09:
I am also experiencing this issue with screen corruption in Marlin mode.

Snargaloo, 2021-12-10:
Here are the config files for the TFT and Marlin:
[TFT35 E3 configs commit #b3322c4.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7689130/TFT35.E3.configs.commit.b3322c4.zip)
[Marlin Bugfix 2.0.x commit #d89f472.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7689131/Marlin.Bugfix.2.0.x.commit.d89f472.zip)

I have a BTT SKR 1.4 Turbo connected to a BTT TFT-35 E3 V3.0 which is connected to an RPi4. I have the RPi4 Bluetooth device disabled and the first PL011 (UART0) is the primary UART. The RPi4 is connected to the TFT from internal GPIO pins on RPi4 to WIFI connector on TFT. The WIFI connector on the TFT automatically echos all data coming to and from the serial connection from the TFT to the SKR 1.4 Turbo.

I only print Through OctoPrint.

radek8, 2021-12-10:
@Snargaloo 
Can you send a photo of the display display damage in simulated Marin mode?

Is it necessary to have step length values to 12 decimal places?
Can't position calculation cause RAM or CPU usage?
Your current settings:
DEFAULT_AXIS_STEPS_PER_UNIT   { 159.543662983178, 159.273390614885, 1593.04916766529, 811.46 }

The Buffer setting is also quite high and can load more RAM
Your current settings:
#define BLOCK_BUFFER_SIZE  8
#define BLOCK_BUFFER_SIZE 64
#define BUFSIZE 32
#define TX_BUFFER_SIZE 256
#define RX_BUFFER_SIZE 1024
#define SERIAL_XON_XOFF

radek8, 2021-12-11:
@Snargaloo
can you try FW Marlin with modified parameters?

#define DEFAULT_AXIS_STEPS_PER_UNIT {159.54, 159.27, 1593.05, 811.46}
#define BLOCK_BUFFER_SIZE 16
#define BLOCK_BUFFER_SIZE 16
#define BUFSIZE 16
#define TX_BUFFER_SIZE 16
//#define RX_BUFFER_SIZE 1024
//#define SERIAL_XON_XOFF

radek8, 2021-12-11:
@NOVAXIM, will you send your configuration?

According to Marlin error messages, large buffer values can cause RAM and CPU overload. Overloading results in display errors, display slowdowns, and eventually printing errors.
https://github.com/MarlinFirmware/Marlin/issues/19477

Snargaloo, 2021-12-11:
Earlier in the year, I was having issues using multi-serial with Marlin.  Working with Marlin developers to debug this issue are the reason for the settings I currently have.  I will send a picture of the screen corruption when it occurs again.  I am going to force some filament run out scenarios.  Each time the corruption occurred previously was during a print that needed the changing of filament.

NOVAXIM, 2021-12-16:
My buffer config was:
```
#define BLOCK_BUFFER_SIZE 16
#define BUFSIZE 4
#define TX_BUFFER_SIZE 0
//#define RX_BUFFER_SIZE 1024
//#define SERIAL_XON_XOFF
```
I changed the settings and soon I will show you the new ones that work. It takes time, be patient.

NOVAXIM, 2021-12-26:
Bad news. Today I started 10 hours printing and after aprox. 3 hours shown first bugs on screen. This happens with latest Marlin 2.0.9.2 with increased buffer (`#define TX_BUFFER_SIZE 32`) and properly updated (without WiFi module). As you know, I have a version for Mega Zero and I have an ESP3D module in connected to the screen board, which caused errors during the firmware update (description is available on the website [ESP3D](https://github.com/luc-github/ESP3D#update). I did a good update and it still shows bugs on the screen. My printer configuration is almost the default, I only changed what I need to do. Previously, I was printing in Marlin mode and there was no problem, but it was all quick prints (max. 1 hour).

I did not notice the printer stuttering after changing the firmware. However, I am surprised by something else, **in the Touch Screen mode it prints faster**, although you cannot see it by looking at the movements. How is this possible?! It is very strange that in Marlin mode it has to be longer. It's the same G code, and I don't see the printer running slower. I will investigate this tomorrow because I have to make a second identical print. Be patient.

**As I can see, the errors are only on the border (they are not on the Marlin content). I did not changed filament, but I stopped print few onces before actually printing.**

NOVAXIM, 2021-12-26:
[image]
I reviewed and in my G code after `M117 Printing...` I do not have any M117 with "E1 Heating..." text. So, why display that?

closed 2022-03-06 by stale[bot]