bigtreetech/BIGTREETECH-TouchScreenFirmware #2277: anyone have a problem when loading the TFT35_E3 V3 "EEPROM CRC mismatch" screen?

askmix, 2021-12-01:
I have tried different SD cards to update, often stops when updating word_unicode.font (59%), the bootloader is updated, I use ready-made firmware... Voltage 5V, 3.3V. When restarting the successfully installed firmware, the EEPROM CRC mismatch window pops up - (stored) 29320 != 46292 (calculated)! EEPROM save and machine Save settings don't help.

askmix, 2021-12-02:
strangely, the CRC error message doesn't pop up today (although the firmware is still the same). Languages English, Russian.

askmix, 2021-12-02:
here's a nuance: crc error pops up when the TFT 35 is connected to Srv2, when the TFT35 is not connected to the mainboard, the screen loads normally. Moreover, when connecting the GND wire and the third pin of the main board of the TFT connector (PA9 "Tx") to the third pin of the RS232 connector (PA3 "Rx") - in "BTT TFT35-E3 V3.0PIN.pdf" for some reason is specified as PA2 "Tx", when I press reset on the main board, a CRC error appears . P.S. I don't have a bootloader on the stm32f407vg. Is that a problem?

askmix, 2021-12-02:
PS where can I find the bootloader for SLRR v2 rev 2?

radek8, 2021-12-02:
This is the storage for the TFT touch screen
Your EEPROM CRC error is a checksum problem on the eeprom of your SKR board. Its size is probably poorly defined.
  Bigtreetech recently began installing these boards with new chips. It is related.
However, your problem is not related to the TFT display. close the ticket and display it in the right place.
Thank you

askmix, 2021-12-03:
I think I found a problem, due to the fact that the Rx Tx lines on STM32f407 and STM32F207 are connected directly, the PA2 port (STM32F207) was damaged (and the date logger shows transmission when the port is unloaded, but even at 15K it cannot control the line, i.e. when connected to the STM32f407 receiver, the line is always 3.3V). Assigned the main port to UART3. now connected to it.

closed 2021-12-03 by askmix

radek8, 2021-12-03:
If you want to use the UART3 connector instead of RS232,
compile the new FW with remapped ports so that you do not have problems with some functions.

radek8, 2021-12-03:
see https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/issues/120

askmix, 2021-12-03:
I didn't use Master firmware, I used "release xx.27" firmware (there is no control of ports 1, 2, 3, 4 and their speeds in the screen settings), I just changed the parameter in the pinTFT35_V3_0.h file to "#define SERIAL_PORT _USART3" // default usart port. So far, I haven't noticed any bugs (the printer is not assembled), but the screen connects, temperatures are transmitted, the reset command from the mainboard comes correctly. Configuration.h did not change. my changes in platform.ini "default_envs = BIGTREE_TFT35_E3_V3_0"

askmix, 2021-12-03:
Поменял те же две строки на мастере, разницы не почувствовал, ошибок не видел (принимает температуру, сохраняет настройки EEPROM, флэшку открывает (но не открывает SD с майнплаты - там скорее всего прошивка SKR2 виновна (запрос идет - ответа нет))

radek8, 2021-12-03:
Use master!