bigtreetech/BIGTREETECH-TouchScreenFirmware #2278: [FR] RRF further tweaks

xPew, 2021-12-01:
Based on my testings I collect some issues/FR's here instead of individual ones,it seems that it worked in the past, so give it a try :)

Fix needed:
- IP/system display in settings/info -> It used to, but for a while its not displaying the IP or the firmware type, it just hangs in a connecting... state.
- Simulation mode dont work (dont trigger the print screen on the TFT) -> I am open to a discussion about this one. PrusaSlicer will enable start simulation after upload, and I can imagine situations when not having this set up is better. Maybe include it in config.ini only for RRF(If memory is a concern, lot of things can be removed from it for RRF which isnt useful for us)?
- With the new 3.4 betas others reported weird behavior in the communication in different modes (S0 1 2 3), so preferably after the stable RRF release, it might need some adjustments as well.

Other(s) FR's:
#2141 
#1979 

I am open for discussion about RRF.

oldman4U, 2021-12-08:
@bttguy what happened to #1791, is it sill active or should we remove it from the Read Me?

bttguy, 2021-12-08:
Well, most of the things that I wanted are got solved, so I closed it. Since than this new list got opened with FR's I didn't thought of (besides the macro menu grid, but yeah, it's brought up here again so...)
You can change my FR to this one to the ReadMe maybe. And if I will need anything besides these I can comment them here.

oldman4U, 2021-12-09:
@xPew - Would it be OK for you to mention this ticket as the main "entrance" for RRF related issues and bug reports, or should we just remove the ticket from the Read Me?

xPew, 2021-12-09:
@oldman4U Yeah, sure, thanks! It can be useful, I will try to look after it.

oldman4U, 2021-12-29:
Those feature requests are still listed in the FR List:

• Menu system for macros + "interactive" macros
• Thumbnail and menu system support for onboard gcodes
• Make the load/unload menu useable

Are they still open?

Pls let me know

Thank you

xPew, 2021-12-30:
Hi!

TLDR, they are all can be removed. Explanation for anybody that is interested:

Menu system and interactive macros are implemented as far as I know, maybe it need further polish, but for the macros I use I don't see any bug, until any request you can remove this.
Thumbnails cant be enabled for us due to the slow transfer speed between the mainboard-TFT, so its not a solvable FR (at least for now hopefully :D )
Also the Load-unload menu is working with a small trick that it execute load.g and unload.g macro what is specific for RRF, rather than the Marlin equivalent Gcode that executes it.

Thank you!

ScottyMakesStuff, 2022-01-06:
I've got three ender 3 Pro RRF builds with a Mellow Fly RFF E3 Pro mainboard and  I'm having problems with the BIGTREETECH TFT35 E3 V3.0 Touch Screens which all have been giving me "Busy processing, please wait..." messages after being on for a short time. I've tried different baud rates and S numbers 0 to 3 and found that 'M575 P1 S2 B57600' is the only line in my config.h file that works. Has anyone here had this issue or have an idea what I can do to fix it?

jaysuk, 2022-01-06:
> ID: <bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2278/1006235633@
> github.com>

It's a known issue. At the moment the only guy who looks at the firmware
hasn't been able too. Issues with it keep getting fixed and then someone
"enhances" something in the firmware and breaks RRF support so it's
something you'll have to get used to being a regular occurrence

ScottyMakesStuff, 2022-01-06:
Hi Jaysuk, thanks for responding. I do appreciate the voluntary efforts of those wonderful people so I can only watch and wait for a potential fix in the future. 
I wonder if BigTreeTech TFT35 E3 actually supports their own Ender 3 RRF board. You'd hope so!

jaysuk, 2022-01-06:
it doesn't. it suffers from the same issue as with any other RRF board.

ScottyMakesStuff, 2022-01-06:
gee whiz, I bought 4 sets of the BTT RRF board and TFT35 kits direct from BTT. That sinks some plans.

jaysuk, 2022-01-06:
Sorry. The best way to keep up to date with whats going on is to join the discord channel for the firmware. It can be found here https://discord.gg/uS97Qs7

ScottyMakesStuff, 2022-01-06:
Thanks for the suggestion, I just joined the discord.

oldman4U, 2022-01-06:
@xPew

Do you see this problem too?

xPew, 2022-01-06:
@oldman4U Yeah, this is kinda what I ment with “ With the new 3.4 betas others reported weird behavior in the communication in different modes (S0 1 2 3), so preferably after the stable RRF release, it might need some adjustments as well.” But it might be caused by what jay mentioned, that some added code messes up RRF.
Best to wait until our programmer will have time, or maybe ask for @bigtreetech or @digant73  if they can help to investigate this. There are a few of us that can help with the testing, sadly not much programmer.

oldman4U, 2022-01-06:
Got it. Thank you

LawnMo, 2022-03-04:
Is there any updates on this ? It's been long enough this product has not been in a proper working state for RRF users. Any plans to investigate / fix issues @oldman4U @bigtreetech @bttguy ?  
I'm getting tired of telling people this firmware is broken and their TFT screens won't work on their fancy printers.  

Any communication would be a step forward in reassuring people are not just wasting money for nothing.

oldman4U, 2022-03-04:
I use Marlin and only try to help organize things.

LawnMo, 2022-03-05:
> I use Marlin and only try to help organize things.

My bad, thought you were part of their "team". Welp, good thing other manufacturers are stepping in.

pfn, 2022-04-30:
I updated my printer using the MKS TFT28 to 3.4.0 and latest TFT firmware, everything seems to work fine for me with `M575 P1 S2 B57600`, S0 and any variant with checksums probably don't work and I'm not inclined to try them atm.

xPew, 2022-06-29:
Dear bot, there are plenty to do to tweak for RRF, so give us some time and wish for the best.

oldman4U, 2022-09-21:
ToDo ;-)

xPew, 2022-09-22:
Yep, still.

oldman4U, 2022-09-22:
👍🏻could you pls do the same with ticket #1830. 

Bot does not listen to me. 

Thank you

xPew, 2022-11-23:
Please @bigtreetech help on this one not to let it stale like this

sinole1987, 2023-02-05:
the auto Z align doesn't work in RRF, is it possible to change the code it runs to G30?