bigtreetech/BIGTREETECH-TouchScreenFirmware #2282: Wifi module working fine in TFT35 v3.0 but not in TFT35 E3 v3.1

NestorUnzu, 2021-12-06:
**Your question or details you need. Please describe.**
I'm setting up a system with a SKR mini E3 v2.1 and a TFT35 touchscreen. 

I connected and configured the wifi module with a TFT35 v3.0 and it's working fine (access point mode, haven't tryied cliente mode), however, if I change the screen to a TFT35 E3 v3.1 my laptop can't find the ESP3D wifi (again, I'm using it in access point mode)

I'm using the latest version of the screen FW downloaded today.

**specify the device variant related to your question.**

SKR mini E3 v2.1
TFT35 v3.0
TFT35 E3 v3.1

**Additional context**
With the "rectangular" TFT35 v3.0 the wifi works fine, with the "square" TFT35 E3 v3.1 the wifi is not working.

Both screens have the same configuration files, the only difference between them is the .bin file.

radek8, 2021-12-06:
Is the TFT 35 E3 v3.1 display new or is it an error?
I have a TFT35 E3 V3.0 display, an EPS01S module connected in client mode and it works fine.
Do you have the same speeds set on the serial port of the display and EPS module?

oldman4U, 2021-12-06:
I use the same configuration like radek8 also without any issue. 3.1 sounds interesting....

xPew, 2021-12-07:
Well, its a miracle from [Two Trees](https://www.aliexpress.com/item/1005003202609788.html?spm=a2g0o.productlist.0.0.38f659cdi1XoZ5&algo_pvid=9d9739bd-3291-4331-988e-0c5d3915cd6a&algo_exp_id=9d9739bd-3291-4331-988e-0c5d3915cd6a-30&pdp_ext_f=%7B%22sku_id%22%3A%2212000024639242320%22%7D) I guess
They call everything they clone from BTT the same +0.1 lol

NestorUnzu, 2021-12-07:
So I bought a chinese clon from a chinese product, never heard of that before.

Guess I'll have to stick to the TFT35 E3 v3.0 and pay more attention to what I'm buying in the future.

I'm checking the baudrates in the two trees LCD board just in case.

oldman4U, 2021-12-07:
And the pin assignment.

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2021-12-08 by NestorUnzu