bigtreetech/BIGTREETECH-TouchScreenFirmware #2283: [BUG] (short description)

jel111, 2021-12-06:
### Description
"illegal flash"
When updating firmware a micro SD adapter doesn't work. Have to use a Full size SD card.

### Steps to reproduce
Tried using a micro SD card with adapter and kept getting Illegal flash. Most times it doesn't work but every so often it does. It seems like the first time it worked but then didn't after that.

**Expected behavior**
Use any SD card especially a micro SD since that's pretty much what everyone has nowadays. At least people I know.

**Actual behavior**
Can only use a full size SD card when not using this you get the "illegal flash"

### Hardware Variant
TFT35 V3.0 E3
### TFT Firmware Version & Main Board Firmware details
/BIGTREE_TFT35_V3.0_E3.27.x.bin

### Additional Information
Just wanted to put this here as I searched for quite awhile until I found a subReddit that discussed this.

radek8, 2021-12-06:
I have the same HW as you. I use a micro SD card with an adapter and no problem.
The problem will most likely be in your card.
What card do you use and what format?
You should use a card up to 8GB or smaller formatted as FAT32
If you use a larger card or a format other than FAT32, you may have problems, and it doesn't matter if you use an SD card or a micro SD adapter.

oldman4U, 2021-12-06:
Please update the title of this ticket and close it after that. Your problem is not firmware related. Thank you

gordo3di, 2021-12-06:
@jel111 

> I have the same HW as you. I use a micro SD card with an adapter and no problem. The problem will most likely be in your card. What card do you use and what format? You should use a card up to 8GB [...]

I have a handful of bad adapters so if I've had any issue it was either the adapter, card formatting or simply a bad board (we've had 8 bad boards and counting thanks bigtreetech...)

radry, 2021-12-06:
Seems more like a hardware problem, not something software can fix. A micro SD adapter only routes the pins from the smaller card to a larger card, there is no circuit inside.  

[image]

Try a different adapter.

xPew, 2021-12-07:
I also only used micro sd cards with an adapter, and I updated the TFT pretty much every week in the past half or one year as new features added, and never had any issues with them.
I even 16GB ones, and maybe 32 as well since I have these around on my desk to work with randomly…

oldman4U, 2021-12-11:
@Msq001

Please close this ticket.

jel111, 2021-12-11:
I've had cards fail before so I tried a few different ones with different adapters. I usually try to keep the adapter with the particular card it came with. 
I also tried an 8 gb card as I had seen someone post about that as well.
The only thing that worked was a full size card that was 32gb I could be wrong but maybe there is a hardware issue with some units and not others?

oldman4U, 2021-12-12:
And you understand that this is a firmware repository and not a hardware related user group? And that the title of the ticket is not related at all.

digant73, 2021-12-14:
@jel111 clean the SD slot in the TFT with a brush and alcohol. Also do not push too much the SD adapter in the TFT. Try to pull a little bit the adapter after it was inserted in the TFT

jel111, 2021-12-15:
> And you understand that this is a firmware repository and not a hardware related user group? And that the title of the ticket is not related at all.

It is related in a second cousin type of way. If you are trying to get it working you would think it a firmware issue. I posted it because I had a hard time finding an answer and only luckily stumbled upon a post. If it's wrong then remove it.

oldman4U, 2021-12-15:
Only the owner of the ticket and the repository can change the title and the status.

closed 2022-01-05 by bigtreetech