bigtreetech/BIGTREETECH-TouchScreenFirmware #2288: [BUG] Using the e-step tuning with an IDEX setup will always use Extruder 1 (T0) and not Extruder 2 (T1)

HappyHound, 2021-12-15:
### Description

I have an Ender 3 Pro with the Ender IDEX motherboard, dual extruders etc. I also have the TFT35 E3. I've installed the 3.27.0 firmware and have updated the e-stop on extruder 1 (T0) with the (very nice) e-step "wizard". But when I select extruder 2 (T1) and press the "Extrude 100mm", it selects says active extruder T0 and not T1. So there seems to be a bug there?
### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details
TFT35 E3 V3 -  BIGTREE_TFT35_V3.0_E3.27.x

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

radek8, 2021-12-19:
This is not a direct mistake. This feature was created by @traffic-light for a printer with only one extruder.
Unfortunately, traffic-light has been offline for some time.
Maybe one of the current active programmers could look at this feature and rework it for more extruders.

digant73, 2021-12-20:
fixed in #2286

closed 2021-12-24 by bigtreetech