bigtreetech/BIGTREETECH-TouchScreenFirmware #2290: [FR] (language  PT-BR in default build)

Marcos2BR, 2021-12-17:
Could you add the BRAZILIAN PORTUGUESE language in the default build? 

I used the Portuguese PT base, and translated most of the texts into Brazilian Portuguese, but I still have to use the language_pt.ini file (see more in attachment). 

the translation needs a minor revision, but it already works with the new BIN (december 16 2021) that works neopixel event + SD card

thanks 
[language_ptbr.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7735554/language_ptbr.zip)

oldman4U, 2021-12-17:
This is open source. Make a PR which includes what you would like to change/add.

radek8, 2021-12-17:
@oldman4U 
Adding a new language is not just adding one INI file.
The language must also be added to the program. Not everyone is a programmer

@Marcos2BR 
 If anyone has to add it, you have to translate the file
\ BIGTREETECH-TouchScreenFirmware \ TFT \ src \ User \ API \ Language \ language_XX.h
the ini file is generated automatically from this file.

Marcos2BR, 2021-12-17:
> @oldman4U Adding a new language is not just adding one INI file. The language must also be added to the program. Not everyone is a programmer
>
> @Marcos2BR If anyone has to add it, you have to [...]

I understood,
  I have now seen the : ... API / language / language_pt.h,
The translation is not complete either, there are many English words yet.
Are you interested in implementing PTBR language? can i help

radek8, 2021-12-17:
Is there a big difference between Portuguese and Brazilian Portuguese?

oldman4U, 2021-12-17:
Not saying that it is easy but maybe he knows!?

Marcos2BR, 2021-12-18:
> Is there a big difference between Portuguese and Brazilian Portuguese? Can't use standard Portuguese?

little difference PT-PT / PT-BR
but Portuguese PT has very little translation, almost everything is in English.

radek8, 2021-12-18:
So if you add the Portuguese translation to the language_pt.h file and send it to me, I will create a PR.

Marcos2BR, 2021-12-18:
> So if you add the Portuguese translation to the language_pt.h file and send it to me, I will create a PR.

Portuguese brasil(PT-BR) is a little different from Portuguese from Portugal(PT-PT) .
have words and expressions that I don't know how to translate to PT-PT

radek8, 2021-12-19:
So create a translation in Brazilian Portuguese.
Maybe someone will take over and implement your translation.
If this language is also the official language, you may want to name the file language_br.h

radek8, 2021-12-19:
I have a PR ready, if you translate the attached file, I will submit it for merging.
[language_br.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7741371/language_br.zip)

radek8, 2021-12-20:
In translation, try to be concise and concise.
Long translations may not fit all on the screen and an inappropriately abbreviated title may be incomprehensible ...

radek8, 2021-12-27:
Send a translation or close the ticket.

Marcos2BR, 2021-12-27:
> Send a translation or close the ticket.

good morning, I've done a good part of the translation, about ten strings to go. but I need to proofread (text size) . I'm comparing with Italian to know the size. I'm on vacation, I'll be back in January. Let's close this tread, in January I'll open one with the translation ready ok?

radek8, 2021-12-27:
OK
leave open until January ...

Marcos2BR, 2022-01-20:
Voltei , desculpem a demora.
arquivo "language_br.h" traduzido , 
falta pequena revisão, vou subir o arquivo aqui porquer amigos do grupo do ZAP e face vão ajudar a revisar

I am back 
sorry for the delay.
The file "language_br.h" its translated,
It need a little review, I'm going to upload the file here because friends from the ZAP group and face will help to review 

[language_br_V1.0.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7905451/language_br_V1.0.zip)

radek8, 2022-01-20:
OK
I'll create a PR
Later changes after testing are possible.

radek8, 2022-01-20:
I fixed the gaps
[language_br.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7907461/language_br.zip)

closed 2022-01-25 by bigtreetech

radek8, 2022-01-25:
Merged.
@Marcos2BR 
It would be good if you learned to work in GitHub and continued to take care of your language and translate new terms and offers that will come in future updates.
Unfortunately, a lot of translations ended with the first addition and no one has updated them since. They have barely 50% translated and it is a pity. I'm not talking about the more translations, the more work is done by programmers who implement new features.

Marcos2BR, 2022-01-27:
hello, language_br.ini from Commits on Jan 25, 2022 does not have the string "label_event_led" , so it doesn't work

I improved the translation of words a bit in language_br.h 

attach the language_br.h ZIP 

[language_br_V1.1.zip Jan 25, 2022](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7946776/language_br_V1.1.zip)

_**PS: FOR EVERYONE, who can help improve the translation, the community thanks **_

radek8, 2022-01-27:
OK, I'll fix it tonight.

radek8, 2022-01-27:
Corrected, added PR #2359

PR #2336 (2022-01-25): Addition of Brazilian Portuguese
Addition of a translation into Brazilian Portuguese

resolves #2290