bigtreetech/BIGTREETECH-TouchScreenFirmware #2296: [BUG] (short description)

wd5gnr, 2021-12-22:
I see this was already posted as an issue and then closed with no resolution.

I have a TFT 35 3.0. TFT35_V3.0.27.x Dec 14 2021 with updated config.ini.

I have UART 3 turned on at 250K baud connected to a Raspberry Pi. If I use picoterm I can send data and it works like you think (e.g., G28 homes, M115 reports, etc.).

However, Octoprint won't connect -- sort of. It actually connects and you can see all the traffic. But it doesn't get acknowledgments so it "hangs". If you open the advanced section of the terminal and press Fake Acknowledge repeatedly it will dutifully send the commands you've queued up. So the comms are working but you don't get a response back so it waits forever unless you fake it.

The only difference I can see is that Octo puts line numbers and checksums on each line whereas G28 in picoterm is just a plain command.

I hear of people using this combination so there must be either something I have set up wrong or a regression?

oldman4U, 2021-12-22:
Hi.

Please read the ReadMe which you can find at the main page of this TFT firmware repository.

Update the title after that to something related if you still believe that there is a bug in the firmware or something is missing.

Otherwise help the community and close the ticket once you do not need it anymore. Thank you

wd5gnr, 2021-12-22:
Yeah I realized I forgot to title it and fixed that 3 minutes before you posted. Sorry. I've read the Readme and I've also read parseAck.c to try to figure out what's going on. The fact that some have it working makes me think it may be a regression.

digant73, 2021-12-27:
but are the commands sent by Octoprint executed by the mainboard?

wd5gnr, 2021-12-27:
Sort of. If I send a command from the terminal, yes. Octoprint gets stuck waiting for an ACK. So if you go do fake ack then each time you do Octoprint will release another command it has stacked up. So you can do something like:
<press home> (nothing happens)
Go to terminal
<press Fake Ack>
(printer homes)

So yes, if you can coax Octoprint into sending the command it is executed.

closed 2022-02-15 by bigtreetech

wd5gnr, 2022-10-02:
In case anyone is revisiting this... since the issue was closed, I decided to try again. I still couldn't get it to work reliably at first. However, I realized this time I had Meatpack turned on in Octoprint. I don't think it was on before and it never worked before. This time, it would work after a reboot (because the Meatpack protocol wasn't on yet). But if Octoprint had been connected and then reconnected, it would fail with no ACK. The solution, of course, is to quit using Meatpack.

Just thought I'd drop that here in case it helps someone else. I know, I know, it seems obvious now, but I hadn't really thought about that being turned on. It is also surprising that Octoprint doesn't issue some sort of hello command that takes that into account. The hello command was just getting sucked in and TFT never returns an OK. Oddly enough, from a terminal it would work, though. Not sure why that is.