bigtreetech/BIGTREETECH-TouchScreenFirmware #2300: [BUG] Thumbnail at Print Screen - Babystep Button doesn't replaced with part thumbnail as icon

sarvenn, 2021-12-26:
### Description
Part thumbnail view was used to be seen at babystep button while printing. I used to use PrusaSlicer 2.3.3 and post processing script which worked awsome.
I have updated to PS 2.4.0 Beta1 (final release does not start on my PC). Current post processing script does not work on PS 2.4, so I enabled Base64 in my TFT35 firmware via configuration.h.
Without using a post processing script, default 95x95 PS thumbnail works ok, but when I start to print, I see default babystep icon.

Another issue is that, when I opne my SD card directory, I see thumbnail, which are OK, I switch to second page of my directory, thumbnails are still ok, but when I switch to first page again, thumbnails are not visible anymore, switch to second page, same thing happens, thumbnails are not visible anymore. When I go back up and enter to SD card directory again, thumbnails are visible again.

My gcodes processed with the post processing script for thumbnail does not have any issue. When I start to pring they are all replaces the babystep icon in printing screen. Besides that they do not dissappear when I switch between part directory pages.

### Hardware Variant
TFT35 E3 V3
### TFT Firmware Version & Main Board Firmware details
TFT35->25 Dec 2021 fw
Marlin Bugfix2

oldman4U, 2021-12-26:
You know that this is related to the new slicer version #1238.

So you started this ticket to let others know or...??

sarvenn, 2021-12-26:
No, these are two different issues.
The one you mentioned above is about the post processing script compatibability with new prusaslicer. I solved it with enabling base64 in TFT fw ( configuration.h). But this time icons doesn't replace the baby step icon now. Older post processing script gcodes have no issue. There replace the babystep icon in print menu.
But this issue is all about thumbnail replacing the babystep icon in print menu.

digant73, 2021-12-27:
@sarvenn verify if PR #2292 fixes your problem

sarvenn, 2021-12-27:
> @sarvenn verify if PR #2292 fixes your problem

This issue still exist with that PR:
_Another issue is that, when I open my SD card directory, I see thumbnail, which are OK, I switch to second page of my directory, thumbnails are still ok, but when I switch to first page again, thumbnails are not visible anymore, switch to second page, same thing happens, thumbnails are not visible anymore. When I go back up and enter to SD card directory again, thumbnails are visible again._

I switched to thumbnail parse 0 (classic) and I use standalone pyhton script at my PS 2.4 beta1 slicer. With this setup in printing menu, babystep icon is replaced with part thumbnail.

Since I changed my thumbnail parser I can not check if base64 png mode still have the issue.
My issue had happened with thumbnail parser 2 (base64 png).

sarvenn, 2021-12-27:
I think page or icons need to be refreshed to view the thumbnail icon while changing the pages. If not, then the icon is replaced with default gcode file icon. When I click on back and enter to my folder again, icons are visible again, or when I click on a gcode file and click on cancel on the printing prompt pop up, page is refreshed and thumbnail icons are alive again.

DynaMight1124, 2021-12-30:
Theres deffo a bug with the thumbnail icon's when going back pages, not sure when it started. I previously had a version I compiled end of Oct with was able to cycle through the pages without any issues (I had a mix of thumbnails from the script and builtin prusaslicer). Just flashed the latest version right now and I have the same issue as above.

I also use Prusaslicer and had previously compiled with base64 png but I have flashed with the classic parser now since the standalone script has been updated.

sarvenn, 2021-12-31:
> Theres deffo a bug with the thumbnail icon's when going back pages, not sure when it started. I previously had a version I compiled end of Oct with was able to cycle through the pages without any [...]

I have experienced same as you @DynaMight1124. I also used to be changing the pages in the past without any issue.

sarvenn, 2021-12-31:
> > Theres deffo a bug with the thumbnail icon's when going back pages, not sure when it started. I previously had a version I compiled end of Oct with was able to cycle through the pages without any issues (I had a mix of thumbnails from the script and builtin prusaslicer). Just flashed the latest version right now and I have the same issue as above.
> > I also use Prusaslicer and had previously compiled with base64 png but I have flashed with the classic parser now since the standalone script has been updated.
> 
> I have experienced same as you @DynaMight1124. I also used to be changing the pages in the past without any issue. I'm not sure but I guess my previous fw that used to be working was released in early December.

sarvenn, 2022-01-01:
@digant73 @guruathwal  hello guys,
could you check the video that @DynaMight1124 shared above. This is what I encountered exactly.
I used to use an earlier fw release which had no issue, I think I was an early December released, may be late November realesed.

guruathwal, 2022-01-01:
@sarvenn can you provide me with the g-code files with the thumbnails?

digant73, 2022-01-01:
@guruathwal is it also possible that bug #2304 needs a fix similar to #2192? maybe it could be useful to make Terminal more robust to avoid any infinite loop

sarvenn, 2022-01-01:
[Ataturk_0.2mm_52m_PLA_Microzey_Siyah_btt.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/7797669/Ataturk_0.2mm_52m_PLA_Microzey_Siyah_btt.zip)
This is compiled with PrusaSlicer and a post processing script for classis thumbnail parser. I also tried regular prusaslicer gcode with base64 png and TFT fw with thumnail parser base64 png ( mode 2). Same issue happens. First I tought that may be thumbnail parser setting is using too much memory which is also mentioned by @CyberDeck in his PR. So I changed my fw's thumbnail parser mode to 0 again.

guruathwal, 2022-01-01:
@sarvenn I tested with the file you provided with the latest pre-built binaries. I was unable to reproduce the issue seen in the video. Everything seems to work ok and all the thumbnails loaded as expected even after switching between multiple pages both up and down. The thumbnail also loaded successfully on the **Babystep** button while printing.

guruathwal, 2022-01-01:
@digant73 regarding #2304, I tried to reproduce the issue. the G28 and G29 worked fine through the terminal screen.

digant73, 2022-01-01:
@sarvenn I also tested your file (and others) and everything is ok. The thumbnails are properly displayed moved up/down and even back/forth folders. I had THUMBNAIL_PARSER set to 0 in Configuration.h

@guruathwal I also did the same and no issue. Although I have UBL configured in Marlin (and not BBL as the user reporting the issue)

DynaMight1124, 2022-01-01:
Is it because we maybe have a mix of script thumbnails and Prusa made thumbnails.

guruathwal, 2022-01-01:
@digant73 I have BBL configured in mine. and it works fine.

digant73, 2022-01-01:
@guruathwal ok. I also asked user having the issue to reduce the baudrate. He reported 1M. Maybe that causes a buffer corruption

sarvenn, 2022-01-01:
That's weird indeed. Could you do the same trial with thumbnail parser set to 2.
By the way, what is BBL @guruathwal ?

I compile my own TFT FW because I need to uncomment this line => #define MBL_START_Z .
That's all. I haven't tried pre-compiled fw. I can try it to see if it changes something.
I use MBL (nozzle as probe). Nothing special in my Marlin fw. My baudrate is 11500.
It is really wierd. I will re-flash my TFT firmware and do a couple of trials again.
Parser 0 => Post processing script modified g-code
Parser 2 => Default PS g-code outcome

I will check both browsing the pages and print screen babystep icon and let you know guys.
Besides I will upload a late October fw version for comparison. Something must have changed :)

This is not the place to report but another old issue has also revived. While printing something I enter to babystep menu to adjust z offset, whenever a set my perfect z offset and click on save in babystep menu, even the numbers on the screen are correct, my nozzle jumps up to a higher position. I never tried in marlin mode, so I don't know if it is marlin or tft related. This was an old and solved issue in our issue history. I don't think there could be a relation but I just wanted you to know.

sarvenn, 2022-01-02:
I found the root cause @digant73 .
This parameter is causing the thumbnails not to refresh and show the icon. If it is enabled (which is enabled as default), then there is no problem, if it is disabled then this problem occures. In my default config.ini setup it was disabled, that's why I could find this issue. So could you please check this PR #2284 @digant73 ? 

I will use this parameter as enabled from now on, I never had problem with the file extension visibility but when you added this feature I just wanted to use it :).

By the way may I ask you to check the in print babystep adjustment and save to eeprom issue as well, I know this is not the proper place but could you also give it a try and tell me if you could confirm it.

#### Filename Extension
# Display fullname for files listed in List Mode / Icon Mode menu.
# If disabled, any filename extension starting with ".g" or ".G" (e.g. ".g", ".gco", ".gcode" etc.) will be hidden.
#   Options: [disable: 0, enable: 1]
filename_extension:1

sarvenn, 2022-01-02:
@discip has reported this issue 19 days ago. I just noticed that @digant73 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2284#issuecomment-994069146

digant73, 2022-01-02:
@sarvenn ok thanks. I found the bug

sarvenn, 2022-01-02:
> @sarvenn ok thanks. I found the bug

Thank you @digant73 for the PR.
Did you have the chance to look for the live Z (babystep) problem?

digant73, 2022-01-02:
@sarvenn Reopen the bug removing from the list the bug (missing thumbnail moving back and forth on the icon list). I previously marked this bug as fixed by my PR while the main bug you reported (icon on babystep button) seems not fixed yet.

sarvenn, 2022-01-02:
> @sarvenn Reopen the bug removing from the list the bug (missing thumbnail moving back and forth on the icon list). I previously marked this bug as fixed by my PR while the main bug you reported (icon on babystep button) seems not fixed yet.

@digant73 done. I included both in the title.

closed 2022-01-04 by bigtreetech

sarvenn, 2022-01-06:
@bigtreetech , @digant73  I could not check if my second part of bug report ( Babystep Button doesn't replaced with thumbnail) is resolved or not with latest fw updates ( #2308 ). 
@digant73 had requested me to open the issue since babystep button issue was not solved. Can anyone confirm it? Otherwise I may kindly request you to re-open the item @bigtreetech.
I can check it this weekend as well.

gordo3di, 2022-11-08:
I've recently been trying to get the thumbnails to show up again in the printing screen. After two full days of messing around I am no further along. Something is wrong in the ICON_PREVIEW code but I still can't figure out why. I was also trying to get the thumbnail to show up in the start print popup which is what started all this.

In PrintingMenu.c I just forced the model_DirectDisplay to see if I could get it to show up. In the code below it will show for a brief millisecond then is immediatly replaced by the move icon.

```c
    printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_PREVIEW : ICON_BABYSTEP;
    //gCreate testing to make preview thumb work
    //printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_BABYSTEP : model_DirectDisplay(getIconStartPoint(5), infoFile.path);
    printingItems.items[KEY_ICON_5].icon = model_DirectDisplay(getIconStartPoint(5), infoFile.path);
```

Changing it to ICON_PREVIEW will have some icon show up that I've never seen with no brief thumbnail.

```c
    printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_PREVIEW : ICON_BABYSTEP;
    //gCreate testing to make preview thumb work
    //printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_BABYSTEP : model_DirectDisplay(getIconStartPoint(5), infoFile.path);
    printingItems.items[KEY_ICON_5].icon = ICON_PREVIEW;
```
[image]

I was also trying all the different icon and display routines I could find but nothing would make the thumbnail work.
```c
      //gcreate test to see preview icon
      //printIconItems.items[KEY_ICON_1].icon = infoFile.modelIcon;
      //printingItems.items[KEY_ICON_5].icon = (infoFile.source < FS_ONBOARD_MEDIA && isPrintModelIcon()) ? ICON_PREVIEW : ICON_BABYSTEP;
      //ITEM.icon = {ICON_FILE, LABEL_NULL};
      //ICON_NULL.item(ICON_FILE);
      //popupDialog(DIALOG_TYPE_QUESTION, LABEL_PRINT, (uint8_t *)temp_info, LABEL_CONFIRM, LABEL_CANCEL, startPrint, exitFolder, NULL);
      //menuDrawIconOnly(ICON_FILE, 1);
      //GUI_ClearRect(480-95, 320-95, 480, 320);
      //GUI_DrawRect(480-95, 320-95, 480, 320);
      //model_DirectDisplay(getIconStartPoint(i), infoFile.path)
      //menuDrawItem(ICON_FILE, i);
      //IMAGE_ReadDisplay(0, 0, ICON_PREVIEW);
      //model_DirectDisplay(0, infoFile.path);
      //IMAGE_ReadDisplay(0, 0, ICON_ADDR(ICON_PREVIEW));
      //model_DirectDisplay_Base64PNG(0, infoFile.path);
      //ICON_ReadDisplay(480-95, 320-95, infoFile.modelIcon);
```

digant73, 2022-11-09:
@gordo3di it is better you open a new bug report providing also a link to this old and fixed bug. Otherwise only people that replied to this thread can read your bug report

PR #2308 (2022-01-04): Fixed #2300
**BUGFIXES:**
* Fixed a bug reported in #2300: Missing thumbnail icon's when going back/forth pages.
* Fixed #2307

fixes #2307 (bug found and solved by @rondlh)

**PR STATE:** Ready to merge
commit 21edb607 (2022-01-04): Fixed #2300 & #2307 (#2308)